MRPT/mrpt #1043: rawlog to rosbag conversion

fafz3958, 2020-04-14:
Hi all, 
Is there any mrpt tool which can convert rawlog to rosbag. There is "mrpt_rawlog" package but currently it can handle odometery and  2D laser scans only. I have a rawlog that contains 3D velodyne data which I want to publish as a ros msg. 
Thanks

jlblancoc, 2020-04-14:
Hi @fafz3958 !
Actually, we have the other way around... apps/rosbag2rawlog in MRPT does the opposite conversion. 
As you said, the correct place for such a tool would be [mrpt_rawlog](https://github.com/mrpt-ros-pkg/mrpt_navigation)... but it doesn't handle that observation type yet. 
Another similar (and unfinished...) project was [mrpt_sensors](http://wiki.ros.org/mrpt_sensors): its aim was to directly publish as ros topics from live sensors, or .rawlog files. 

We are not focused at present on those tools, tough... so if you or someone else wants to help pushing in any of those two projects to provide the feature you need, it would be great to have a pull request for it!

fafz3958, 2020-04-15:
Thank you for your response. Is there a way to export txt files for velodyne scans along with their time stamps from .rawlog. I have tried "rawlog-edit" tool and Rawlogviewer but both export only odometer data but not velodyne scan data. Any provided information is appreciated. Another question is where can I get the related tf data?

jlblancoc, 2022-07-01:
Moved to https://github.com/mrpt-ros-pkg/mrpt_navigation/issues/130

closed 2022-07-01 by jlblancoc