MRPT/mrpt #1044: how can i build 'test_mrpt_maps'??

Songminkee, 2020-04-21:
when i build this proj , I keep getting errors.
how can i download devel.zip?
i used windows 10 , visualstudio 17

downloading 'https://github.com/MRPT/octomap/archive/devel.zip' failed
404 Not found

the8inc, 2020-04-22:
Hi, @Songminkee 

me too.
What your version?

OCTOMAP_EP_URL is writtern at 36 line, in cmakemodules/script_octomap.cmake.

[1.5.x]
download https://github.com/OctoMap/octomap/archive/devel.zip
rename "octomap-devel.zip" to "octomap.zip", and copy to mrpt/otherlibs
[2.0.x]
download https://github.com/OctoMap/octomap/archive/devel.zip
rename "octomap-devel.zip" to "octomap.zip", and copy to mrpt/3rdparty

But, extract-EP_JPEG.cmake:51 cause another error.
`
file RENAME failed to rename
  mrpt/build/3rdparty/ex-EP_JPEG1234/libjpeg-turbo-1.5.90
to
  mrpt/build/3rdparty/libjpeg-turbo
because: No such file or directory
`

Songminkee, 2020-04-24:
Hi, @the8inc 

i downloaded 'mrpt-2.0.1.zip' and use cmake. 

Of course, i re-git clone glfw in 3rdparty/nanogui/ext before start cmake.

then we can get path 'your_cmake_dst_Folder/EP_octomap-prefix/src/EP_octomap-stamp'

and i changed line 114, 78 in download-EP_octomap.cmake

finally, i built 'test_mrpt_maps'.

but i can't build '(EXAMPLE) maps_gridmap3D_simple' solution.

how can i make 'libmrpt-maps201_msvc141_x64-dbg.dll', 'libmrpt-opengl201_msvc141_x64-dbg.dll', libmrpt-obs201_msvc141_x64-dbg.dll.
'libmrpt-gui201_msvc141_x64-dbg.dll'

the8inc, 2020-04-27:
Hi, @Songminkee

I successed to build.
You must to pre-build EP_xxxx.(I use Batch Build.)
After that, build solution.
Don't re-build, because be cleaned EP_xxxx.

Best regards.

[image]

jlblancoc, 2020-04-30:
> I solved this issue.
>I changed 'https://github.com/MRPT/octomap/archive/devel.zip' to >'https://github.com/OctoMap/octomap/archive/devel.zip'.
>but I can't build 'maps_gridmap3D_simple' dut to assimp.

Hmmm... that was fixed 759b561ae2a1780b86b327e3daccb5b48b5ef29a 17 days ago... perhaps you are using a version older than "develop", right? 
I'll anyway release a new stable version soon, so nobody else finds this problem. 

> Of course, i re-git clone glfw in 3rdparty/nanogui/ext before start cmake.

A better alternative is using "git clone" instead of downloading the zip. 

Cheers

closed 2020-04-30 by jlblancoc

Songminkee, 2020-04-30:
Yes, I used an older version(the stable version on homepage).

@the8inc @jlblancoc 
Thank you for responding to this even though it was an inexperienced issue.