MRPT/mrpt #1046: undefined reference to symbol 'pthread_key_delete

zjlooojoe, 2020-04-21:
HI,do you know how to fix this problem?

[ 47%] Linking CXX shared library ../../lib/libmrpt-graphslam.so
/usr/bin/ld: /usr/local/lib/gcc/x86_64-pc-linux-gnu/7.3.0/../../../libglfw3.a(posix_tls.c.o): undefined reference to symbol 'pthread_key_delete@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: 错误：ld 返回 1
libs/graphslam/CMakeFiles/graphslam.dir/build.make:194: recipe for target 'lib/libmrpt-graphslam.so.2.0.2' failed
make[2]: *** [lib/libmrpt-graphslam.so.2.0.2] Error 1
CMakeFiles/Makefile2:3218: recipe for target 'libs/graphslam/CMakeFiles/graphslam.dir/all' failed
make[1]: *** [libs/graphslam/CMakeFiles/graphslam.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

zjlooojoe, 2020-04-21:
After adding a   “set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11  -pthread" )” in the cmakelist , there are another problem...

[ 47%] Linking CXX shared library ../../lib/libmrpt-graphslam.so
/usr/bin/ld: /usr/local/lib/gcc/x86_64-pc-linux-gnu/7.3.0/../../../libglfw3.a(glx_context.c.o): undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
collect2: 错误：ld 返回 1
libs/graphslam/CMakeFiles/graphslam.dir/build.make:194: recipe for target 'lib/libmrpt-graphslam.so.2.0.2' failed
make[2]: *** [lib/libmrpt-graphslam.so.2.0.2] Error 1
CMakeFiles/Makefile2:3220: recipe for target 'libs/graphslam/CMakeFiles/graphslam.dir/all' failed
make[1]: *** [libs/graphslam/CMakeFiles/graphslam.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

zjlooojoe, 2020-04-21:
After recompile the gflw, porblems are solved.
But there are another problem again....

[ 80%] Linking CXX executable ../../bin/ReactiveNav3D-Demo
[ 80%] Built target ReactiveNav3D-Demo
[ 80%] Building CXX object apps/robot-map-gui/CMakeFiles/robot-map-gui.dir/gui/configWidget/CGeneralConfig.cpp.o
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:45:2: 错误：‘AVCodecParameters’ does not name a type; did you mean ‘AVCodecParser’?
  AVCodecParameters* pCodecPars{nullptr};
  ^~~~~~~~~~~~~~~~~
  AVCodecParser
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp: 在成员函数‘bool mrpt::hwdrivers::CFFMPEG_InputStream::openURL(const string&, bool, bool)’中:
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:149:36: 错误：‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
   if (ctx->pFormatCtx->streams[i]->codecpar->codec_type ==
                                    ^~~~~~~~
                                    codec
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:165:7: 错误：‘struct mrpt::hwdrivers::TFFMPEGContext’ has no member named ‘pCodecPars’; did you mean ‘pCodec’?
  ctx->pCodecPars = ctx->pFormatCtx->streams[ctx->videoStream]->codecpar;
       ^~~~~~~~~~
       pCodec
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:165:64: 错误：‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
  ctx->pCodecPars = ctx->pFormatCtx->streams[ctx->videoStream]->codecpar;
                                                                ^~~~~~~~
                                                                codec
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:168:42: 错误：‘struct mrpt::hwdrivers::TFFMPEGContext’ has no member named ‘pCodecPars’; did you mean ‘pCodec’?
  ctx->pCodec = avcodec_find_decoder(ctx->pCodecPars->codec_id);
                                          ^~~~~~~~~~
                                          pCodec
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:188:48: 错误：‘AVStream {aka struct AVStream}’ has no member named ‘codecpar’; did you mean ‘codec’?
    ctx->pFormatCtx->streams[ctx->videoStream]->codecpar))
                                                ^~~~~~~~
                                                codec
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:186:6: 错误：‘avcodec_parameters_to_context’在此作用域中尚未声明
  if (avcodec_parameters_to_context(
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:186:6: 附注：suggested alternative: ‘avcodec_free_context’
  if (avcodec_parameters_to_context(
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      avcodec_free_context
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:224:8: 错误：‘struct mrpt::hwdrivers::TFFMPEGContext’ has no member named ‘pCodecPars’; did you mean ‘pCodec’?
   ctx->pCodecPars->width, ctx->pCodecPars->height, 1);
        ^~~~~~~~~~
        pCodec
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:224:32: 错误：‘struct mrpt::hwdrivers::TFFMPEGContext’ has no member named ‘pCodecPars’; did you mean ‘pCodec’?
   ctx->pCodecPars->width, ctx->pCodecPars->height, 1);
                                ^~~~~~~~~~
                                pCodec
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:240:8: 错误：‘struct mrpt::hwdrivers::TFFMPEGContext’ has no member named ‘pCodecPars’; did you mean ‘pCodec’?
   ctx->pCodecPars->width, ctx->pCodecPars->height, 1);
        ^~~~~~~~~~
        pCodec
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:240:32: 错误：‘struct mrpt::hwdrivers::TFFMPEGContext’ has no member named ‘pCodecPars’; did you mean ‘pCodec’?
   ctx->pCodecPars->width, ctx->pCodecPars->height, 1);
                                ^~~~~~~~~~
                                pCodec
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp: 在成员函数‘bool mrpt::hwdrivers::CFFMPEG_InputStream::retrieveFrame(mrpt::img::CImage&)’中:
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:313:14: 错误：‘avcodec_send_packet’在此作用域中尚未声明
    int ret = avcodec_send_packet(ctx->pCodecCtx, &packet);
              ^~~~~~~~~~~~~~~~~~~
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:313:14: 附注：suggested alternative: ‘av_append_packet’
    int ret = avcodec_send_packet(ctx->pCodecCtx, &packet);
              ^~~~~~~~~~~~~~~~~~~
              av_append_packet
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:323:10: 错误：‘avcodec_receive_frame’在此作用域中尚未声明
    ret = avcodec_receive_frame(ctx->pCodecCtx, ctx->pFrame);
          ^~~~~~~~~~~~~~~~~~~~~
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:323:10: 附注：suggested alternative: ‘avcodec_free_frame’
    ret = avcodec_receive_frame(ctx->pCodecCtx, ctx->pFrame);
          ^~~~~~~~~~~~~~~~~~~~~
          avcodec_free_frame
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:336:32: 错误：‘struct mrpt::hwdrivers::TFFMPEGContext’ has no member named ‘pCodecPars’; did you mean ‘pCodec’?
[... 6 lines cut ...]
          pCodec
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:338:10: 错误：‘struct mrpt::hwdrivers::TFFMPEGContext’ has no member named ‘pCodecPars’; did you mean ‘pCodec’?
     ctx->pCodecPars->width, ctx->pCodecPars->height,
          ^~~~~~~~~~
          pCodec
[... cut ...]
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:359:10: 错误：‘struct mrpt::hwdrivers::TFFMPEGContext’ has no member named ‘pCodecPars’; did you mean ‘pCodec’?
     ctx->pCodecPars->width, ctx->pCodecPars->height,
          ^~~~~~~~~~
          pCodec
/home/zjlooojoe/mrpt/libs/hwdrivers/src/CFFMPEG_InputStream.cpp:359:34: 错误：‘struct mrpt::hwdrivers::TFFMPEGContext’ has no member named ‘pCodecPars’; did you mean ‘pCodec’?
     ctx->pCodecPars->width, ctx->pCodecPars->height,
                                  ^~~~~~~~~~
                                  pCodec

jlblancoc, 2020-04-30:
If you don't need ffmpeg (to parse video files as inputs), you can safely disable it with the CMake variable DISABLE_FFMPEG (under the "advanced" cmake variables) in ccmake or cmake-gui.

jlblancoc, 2020-05-27:
Closing, since I can't find a way to reproduce the build error... Please, reopen if you still have the problem and have instructions on how to replicate it. Thanks.

closed 2020-05-27 by jlblancoc