MRPT/mrpt #1061: compile error with opencv

hoppss, 2020-06-10:
I meet a compile error, my opencv version is 3.4.10, log as following shows:

`
In file included from /home/cloud/project/mrpt/libs/img/src/CImage.cpp:29:0:
/home/cloud/project/mrpt/libs/img/src/CImage.cpp: In member function ‘void mrpt::img::CImage::resize(std::size_t, std::size_t, mrpt::img::TImageChannels, mrpt::img::PixelDepth)’:
/home/cloud/project/mrpt/libs/img/src/CImage.cpp:260:37: error: conversion from ‘cv::Mat’ to non-scalar type ‘IplImage {aka _IplImage}’ requested
   IplImage ipl = cvIplImage(m_impl->img);
/home/cloud/project/mrpt/libs/img/include/mrpt/3rdparty/do_opencv_includes.h:76:24: note: in definition of macro ‘cvIplImage’
 #define cvIplImage(X) (X)
`

closed 2020-06-10 by jlblancoc

hoppss, 2020-06-11:
thanks for your in time help.

Is there similiar problem in other parts with opencv,  for example,
mrpt/libs/vision/src/CVideoFileWriter.cpp:147:28
mrpt/libs/vision/src/checkerboard_ocamcalib_detector.cpp:89:26:

jlblancoc, 2020-06-14:
should be now fixed 100% via #1063...

PR #1062 (2020-06-10): fix build against opencv 3.4.10
Fixes #1061
PR #1063 (2020-06-14): Fix build against opencv 3.x
## Changed apps/libraries

* mrpt-img

Closes #1061