MRPT/mrpt #1066: Debian packages do not update/broken on bionic

s-hemer, 2020-06-17:
I have installed various packages from https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt-stable , i.e. like 
`libmrpt-slam2.0 (= 1:2.0.4~snapshot20200605-1034-git-94c22039-bionic-1)`

but the newest snapshot (= 1:2.0.4~snapshot20200616-0943-git-83ab9a1e-bionic-1) does not want to install because of
`libmrpt-slam2.0 : Depends: libgcc-s1 (>= 3.0) but it is not installable`

Problem is then that the dev package `libmrpt-dev`  recommends  `libmrpt-slam-dev (= 1:2.0.4~snapshot20200616-0943-git-83ab9a1e-bionic-1)`

Downgrading the dev packages is not possible as there is only the latest available.

jlblancoc, 2020-06-18:
Ok, I'm afraid that's because an extra PPA dependency introduced to support xenial builds... (sigh).

Will have to split xenial into its own PPA, I can't see another solution to support it until its EOL.
The problematic dependency has been removed from the PPA, please, do a "sudo apt update" after 12/24h to check if the problem has gone.

PS: As a temporary solution, you could add [this](https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/test) PPA to your system, then I think it should let you upgrade

s-hemer, 2020-06-18:
But as I wrote it's not about Xenial, it's about Bionic and that was working until begin of June. And at least until the point release of Focal, I'd see Bionic a the stable LTS release.

s-hemer, 2020-06-18:
Ok, problem solved. 
Maybe you should remove the 1:2.0.4~snapshot20200618-0713-git-524af0f5-xenial-1 package from https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt-stable  as you have the https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt-stable-xenial for that?!

closed 2020-06-18 by s-hemer

jlblancoc, 2020-06-18:
> Maybe you should remove the 1:2.0.4~snapshot20200618-0713-git-524af0f5-xenial-1 package from https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt-stable as you have the https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt-stable-xenial for that?!

Done, thanks!