MRPT/mrpt #1069: no valid opengpg data found

5l1v3r1, 2020-06-20:
at kali 2020.4 if you want to install it from its own repo, there are so many conflicting packages to remove and i dont wanna prefer this because those other packages are necessary as well. If i disable kali repos and adding original repo with this method, no valid opengpg data found error is appearing and when i try to apt-get update after the command, it freezes and no action at all. i tried several fixes like adding the key : FFC0070C5BC9DECF manually but unfortunately no avail. Can you please help me with that?

sudo add-apt-repository ppa:joseluisblancoc/mrpt-stable
 Mobile Robot Programming Toolkit (MRPT) packages, git branch: `develop`
Homepage: https://www.mrpt.org/
GitHub: https://github.com/MRPT/mrpt/

**NOTE**: Users of Ubuntu 16.04 Xenial needs this alternative PPA:
https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt-stable-xenial

# ==== INSTALL ====
sudo add-apt-repository ppa:joseluisblancoc/mrpt-stable
sudo apt-get update
sudo apt-get install libmrpt-dev mrpt-apps
 More info: https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt-stable
Press [ENTER] to continue or ctrl-c to cancel adding it

gpg: keybox '/tmp/tmpxk5kzz2n/pubring.gpg' created
gpg: /tmp/tmpxk5kzz2n/trustdb.gpg: güvence veritabanı oluşturuldu
gpg: anahtar FFC0070C5BC9DECF: genel anahtar "Launchpad PPA for Jose Luis" alındı
gpg: İşlenmiş toplam miktar: 1
gpg:                       alınan: 1
gpg: geçerli OpenPGP verisi yok

jlblancoc, 2020-07-11:
Perhaps this is because you are trying to add a PPA repository to a distribution that is not Ubuntu? and then you need to add some PPA keys?

5l1v3r1, 2020-07-17:
well i fixed the gpg key issue. both of the distrubutions are debian so you can add any repo or key with proper methods. you can close the thread thank you.

jlblancoc, 2020-07-17:
Great!

closed 2020-07-17 by jlblancoc