MRPT/mrpt #1071: camera-calib  3D view is crashing in ubuntu 20.04

nrasulnrasul, 2020-07-01:
I had installed mrpt on ubuntu 20.04 from repo  ppa:joseluisblancoc/mrpt and try to use camera-calib app.
The app is crashing when I try to open the 3D view

```
(camera-calib:3698): Gdk-ERROR **: 14:16:28.609: The program 'camera-calib' received an X Window System error.
This probably reflects a bug in the program.
The error was ['GLXBadFBConfig'.]
  (Details: serial 3007 error_code 167 request_code 152 (GLX) minor_code 34)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Trace/breakpoint trap (core dumped)
```

[image]

I had installed mrpt using 

```
sudo add-apt-repository ppa:joseluisblancoc/mrpt
sudo apt-get update
sudo apt-get install libmrpt-dev mrpt-apps
```

jlblancoc, 2020-07-02:
Hmmm... perhaps [this SO question & answers](https://stackoverflow.com/questions/39213757/glfw-cannot-create-a-window-glx-failed-to-create-context-glxbadfbconfig) are related to the crash?

nrasulnrasul, 2020-07-03:
Thanks for the reply. It was crashing due to the graphics drivers.

closed 2020-07-11 by jlblancoc