MRPT/mrpt #1075: How to solve "error: no type named 'Int32' in namespace       'testing::internal'"

Gatsby23, 2020-07-31:
[image]

Hello, My computer is OSX10.15.6 and how to solve this ?

jlblancoc, 2020-08-01:
Hi, 

Per https://github.com/google/googletest/issues/1519 , it seems it's related to c++17 flags. 
What's your version of cmake? 

Anyway, give it a try to the code in the branch [fix/cpp17-gtest](https://github.com/MRPT/mrpt/tree/fix/cpp17-gtest) I just pushed, and let me know if that fixes the problem.

Gatsby23, 2020-08-01:
[image]
My CMake is 3.18.0, but it also fails in fix/cpp17-gtes this branch.

jlblancoc, 2020-08-02:
What version of clang?

danieldanciu, 2020-08-06:
I get the same error. 
OS: Catalina 10.15.3
clang version 9.0.0
cmake 3.15.4

Error:

```
/usr/local/include/gmock/internal/gmock-internal-utils.h:145:21: error: use of undeclared identifier 'Int64'
GMOCK_DECLARE_KIND_(Int64, kInteger);
                    ^
/usr/local/include/gmock/internal/gmock-internal-utils.h:146:21: error: use of undeclared identifier 'UInt64'
GMOCK_DECLARE_KIND_(UInt64, kInteger);
                    ^
In file included from /Users/dd/work/metagenome/metagraph/tests/annotation/row_diff/test_row_diff.cpp:2:
In file included from /usr/local/include/gmock/gmock.h:59:
/usr/local/include/gmock/gmock-actions.h:165:46: error: use of undeclared identifier 'UInt64'
GMOCK_DEFINE_DEFAULT_ACTION_FOR_RETURN_TYPE_(UInt64, 0);
                                             ^
/usr/local/include/gmock/gmock-actions.h:166:46: error: use of undeclared identifier 'Int64'
GMOCK_DEFINE_DEFAULT_ACTION_FOR_RETURN_TYPE_(Int64, 0);
                                             ^
In file included from /Users/dd/work/metagenome/metagraph/tests/annotation/row_diff/test_row_diff.cpp:2:
/usr/local/include/gmock/gmock.h:76:1: error: no type named 'Int32' in namespace 'testing::internal'
GMOCK_DECLARE_int32_(default_mock_behavior);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/gmock/internal/gmock-port.h:73:44: note: expanded from macro 'GMOCK_DECLARE_int32_'
    extern GTEST_API_ ::testing::internal::Int32 GMOCK_FLAG(name)
```

danieldanciu, 2020-08-06:
A quick hack to work around the issue is to define
```
namespace testing {
namespace internal {
using UInt64 = uint64_t;
using Int64 = int64_t;
using Int32 = int32_t;
} // namespace internal
} // namespace testing
```

before including `gmock.h`

Gatsby23, 2020-08-06:
> What version of clang?

[image]
My Clang version is 10.0

jlblancoc, 2020-08-06:
Our CI uses Xcode-11.5 with C++ version (I guess it's apple clang?) version 11... 
It seems a bug in clang 9, right? Will try to include the patch proposed by @danieldanciu in a PR anyway...

jlblancoc, 2021-02-15:
Will close due to the lack of follow-up. 

Please, reopen if it's still an issue with up-to-date OSX compilers.

closed 2021-02-15 by jlblancoc