MRPT/mrpt #1076: mrpt2 failing to build in ROS Noetic

sloretz, 2020-08-03:
It looks like all builds of `mrpt2` are failing in ROS Noetic

* http://build.ros.org/job/Nbin_uF64__mrpt2__ubuntu_focal_amd64__binary/
* http://build.ros.org/job/Nbin_db_dB64__mrpt2__debian_buster_amd64__binary/
* http://build.ros.org/job/Nbin_ufhf_uFhf__mrpt2__ubuntu_focal_armhf__binary/
* http://build.ros.org/job/Nbin_ufv8_uFv8__mrpt2__ubuntu_focal_arm64__binary/
* http://build.ros.org/job/Nbin_dbv8_dBv8__mrpt2__debian_buster_arm64__binary/

It seems like there's a long list of errors about being unable to find libraries on amd64

* http://build.ros.org/job/Nbin_uF64__mrpt2__ubuntu_focal_amd64__binary/
* http://build.ros.org/job/Nbin_db_dB64__mrpt2__debian_buster_amd64__binary/

```
17:55:35 dpkg-shlibdeps: error: cannot find library libmrpt-hwdrivers.so.2.0 needed by debian/ros-noetic-mrpt2/opt/ros/noetic/bin/GridmapNavSimul (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
17:55:35 dpkg-shlibdeps: error: cannot find library libmrpt-kinematics.so.2.0 needed by debian/ros-noetic-mrpt2/opt/ros/noetic/bin/GridmapNavSimul (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
17:55:35 dpkg-shlibdeps: error: cannot find library libmrpt-maps.so.2.0 needed by debian/ros-noetic-mrpt2/opt/ros/noetic/bin/GridmapNavSimul (ELF format: 'elf64-x86-64' abi: '0201003e00000000'; RPATH: '')
...
```

It looks like the build times out on armhf and arm64

* http://build.ros.org/job/Nbin_ufhf_uFhf__mrpt2__ubuntu_focal_armhf__binary/
* http://build.ros.org/job/Nbin_ufv8_uFv8__mrpt2__ubuntu_focal_arm64__binary/
* http://build.ros.org/job/Nbin_dbv8_dBv8__mrpt2__debian_buster_arm64__binary/

```
04:22:31 Build timed out (after 720 minutes). Marking the build as failed.
```

jlblancoc, 2020-08-03:
Hi, thanks for the feedback! 

I know... some hints were reported in [this ROS answers post](https://answers.ros.org/question/356962/should-cmake_install_libdir-be-ignored-in-rep136-packages/), but got no answers yet...

sloretz, 2020-08-03:
I have no clue. Maybe this fix you made for mrpt1 applies here too? https://github.com/mrpt-ros-pkg-release/mrpt1-release/commit/17361d75ef9e73c3496dca74813c4b9ee90ec643#diff-d96281e61c31f35df56361434301b945R98

I wonder where `CMAKE_INSTALL_LIBDIR` is being set.

jlblancoc, 2020-08-04:
@sloretz awesome! Thanks, I totally forgot about that change :-)
It would still be great to get down to the real cmake + LIBDIR + catkin issue, but a workaround allows this to move forward.

jlblancoc, 2020-08-04:
Fixed via a bunch of commits like e.g. https://github.com/mrpt-ros-pkg-release/mrpt2-release/commit/92bfc477af255f46b5b2130c9f88d867de97e3d8

Will leave this open to remember to bloom_release after v2.0.5

jlblancoc, 2020-08-31:
Done here: https://github.com/ros/rosdistro/pull/26382

Thanks!

closed 2020-08-31 by jlblancoc