MRPT/mrpt #1078: missing `eigen_plugins` and `utils` header files

christian-rauch, 2020-08-05:
I am trying to use MRPT on Ubuntu 20.04. Since MRPT is not part of the official Ubuntu 20.04 repo, I have to use the ppa `ppa:joseluisblancoc/mrpt-stable`.

It seems that there are some header files missing in this repo and the ppa. E.g. the upstream package in Ubuntu used to have header files `mrpt/base/include/mrpt/math/eigen_plugins.h` and `mrpt/base/include/mrpt/utils.h`. These seem to be missing in this repo.

Were they removed recently? Is there a way to retain backward compatibility with the old API or is there some transition guide on how to replace old API? E.g. what is the alternative to `getRowCount()` that was part of `eigen_plugins.h`?

jlblancoc, 2020-08-05:
Hi @christian-rauch 

Yes, that API was changed in the transition between mrpt 1.x to 2.x. 
I'm sorry, I dropped support for the mrpt 1.x series in modern versions of Ubuntu due to all the build problems that it caused... just can't afford duplicating the effort, and it's not worth it :-) 

Anyway, here's a transition guide can be reached [here](https://docs.mrpt.org/reference/develop/page_porting_mrpt2.html).
I'm afraid there's not too much info on the particular point you ask: probably the easiest place to look is at [Eigen's docs](https://eigen.tuxfamily.org/dox/group__TutorialMatrixClass.html). For example: `getRowCount()` -> `rows()`, etc.

jlblancoc, 2020-08-16:
Closing, reopen if need it.

closed 2020-08-16 by jlblancoc