MRPT/mrpt #1089: MPRT dependency on ROS Kinetic

brunoeducsantos, 2020-08-25:
Hi,

I am adding dependency on CMake as follows:
```
find_package( MRPT REQUIRED maps slam)
...
target_link_libraries(${PROJECT_NAME}_node
   ${catkin_LIBRARIES}
   ${OpenMP_LIBS}
   ${OpenCV_LIBRARIES}
   ${OCTOMAP_LIBRARIES}
   ${MRPT_LIBRARIES}
 )
```
My system has the following properties:
* Ubuntu 16.04
* CMake version: VERSION 2.8.3
* MPRT installed using current develop branch 

I am obtaining the following error:

>  fatal error: mrpt/slam.h: No such file or directory
>  #include <mrpt/slam.h>         // Include all classes in mrpt-slam and its dependencies
>           ^~~~~~~~~~~~~
> compilation terminated.

Could it be some issue with my current MPRT version?

Thanks,
Bruno

jlblancoc, 2020-08-25:
Hmm... 

Add the following to your cmake file to find out a bit more on what's going on: 

```cmake
find_package( MRPT REQUIRED maps slam)
message(STATUS "MRPT version: ${MRPT_VERSION}")
message(STATUS "MRPT_LIBRARIES: ${MRPT_LIBRARIES}")
...
```

also, while building (e.g. `cd  ~/catkin_ws/build/your-ros-pkg/  && VERBOSE=1 make -j1`) check out for the `-I` flags that should be pointing to MRPT header directories.

brunoeducsantos, 2020-08-26:
This is the resulting log:

>  -DDISABLE_OPENNI2 -DDISABLE_PCAP -DDISABLE_PNG -DEIGEN_DENSEBASE_PLUGIN=\"grid_map_core/eigen_plugins/DenseBasePlugin.hpp\" -DEIGEN_FUNCTORS_PLUGIN=\"grid_map_core/eigen_plugins/FunctorsPlugin.hpp\" -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"local_mapping\" -DvtkFiltersFlowPaths_AUTOINIT="1(vtkFiltersParallelFlowPaths)" -DvtkIOExodus_AUTOINIT="1(vtkIOParallelExodus)" -DvtkIOGeometry_AUTOINIT="1(vtkIOMPIParallel)" -DvtkIOImage_AUTOINIT="1(vtkIOMPIImage)" -DvtkIOSQL_AUTOINIT="2(vtkIOMySQL,vtkIOPostgreSQL)" -DvtkRenderingContext2D_AUTOINIT="1(vtkRenderingContextOpenGL)" -DvtkRenderingCore_AUTOINIT="4(vtkInteractionStyle,vtkRenderingFreeType,vtkRenderingFreeTypeOpenGL,vtkRenderingOpenGL)" -DvtkRenderingFreeType_AUTOINIT="2(vtkRenderingFreeTypeFontConfig,vtkRenderingMatplotlib)" -DvtkRenderingLIC_AUTOINIT="1(vtkRenderingParallelLIC)" -DvtkRenderingVolume_AUTOINIT="1(vtkRenderingVolumeOpenGL)" -I/home/bruno/grid_ws/devel/include -I/usr/include/vtk-6.2 -I/usr/include/freetype2 -I/usr/include/x86_64-linux-gnu/freetype2 -I/usr/include/jsoncpp -I/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent -I/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include -I/usr/lib/openmpi/include -I/usr/lib/openmpi/include/openmpi -I/usr/include/python2.7 -I/usr/include/hdf5/openmpi -I/usr/include/libxml2 -I/usr/include/tcl -I/home/bruno/grid_ws/src/skimap_path_planning/include -I/home/bruno/grid_ws/src/skimap_ros/include -I/home/bruno/grid_ws/src/grid_map/grid_map_ros/include -I/home/bruno/grid_ws/src/grid_map/grid_map_core/include -I/home/bruno/grid_ws/src/grid_map/grid_map_cv/include -I/home/bruno/grid_ws/src/grid_map/grid_map_octomap/include -I/home/bruno/grid_ws/src/grid_map/grid_map_pcl/include -I/opt/ros/kinetic/include -I/opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I/usr/include/eigen3 -I/usr/include/pcl-1.7 -I/usr/include/ni -I/usr/local/include/eigen3 -I/home/bruno/grid_ws/src/skimap_path_planning/include_directories -I"/home/bruno/grid_ws/src/skimap_path_planning/(" -I"/home/bruno/grid_ws/src/skimap_path_planning/)" -isystem /opt/ros/kinetic/include/opencv-3.3.1-dev -isystem /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv  -fopenmp -pthread -O3 -DNDEBUG   -std=c++11 -o CMakeFiles/map_server.dir/src/map_server.cpp.o -c /home/bruno/grid_ws/src/skimap_path_planning/src/map_server.cpp

When installing from the source doing `sudo make install`, I got the following partial log:

> -- Up-to-date: /usr/local/include/mrpt/typemeta/include//mrpt
> -- Up-to-date: /usr/local/include/mrpt/typemeta/include//mrpt/typemeta
> -- Up-to-date: /usr/local/include/mrpt/typemeta/include//mrpt/typemeta/xassert.h
> -- Up-to-date: /usr/local/include/mrpt/typemeta/include//mrpt/typemeta/TEnumType.h
> -- Up-to-date: /usr/local/include/mrpt/typemeta/include//mrpt/typemeta/TTypeName.h
> -- Up-to-date: /usr/local/include/mrpt/typemeta/include//mrpt/typemeta/num_to_string.h
> -- Up-to-date: /usr/local/include/mrpt/typemeta/include//mrpt/typemeta/static_string.h
> -- Up-to-date: /usr/local/include/mrpt/typemeta/include//mrpt/typemeta/TTypeName_stl.h
> 

Do you think that could be path issue to include dirs ?
And maybe the libs should be  on `/usr/local`  instead of `/usr/share` for ROS accessing it ?

Looking forward to your feedback,
Bruno

brunoeducsantos, 2020-08-26:
I sorted the issue. It was a rookie mistake. Forget to add one `${MRPT_LIBRARIES}` on my `target_link_libraries`.
Sorry for the trouble.

closed 2020-08-26 by brunoeducsantos