MRPT/mrpt #1093: COccuppancy grid initialization

brunoeducsantos, 2020-09-17:
Hi @jlblancoc ,
Thanks for the amazing work.
I am having an issue with occupancy grid initialization. Maybe it is a simple one.

``` 
mrpt::maps::COccupancyGridMap2D::Ptr map(new mrpt::maps::COccupancyGridMap2D(left_corner(0), left_corner(1), upper_corner(0), upper_corner(1), grid_map.getResolution()));
```
and I am getting the following issue:
```
C++ exception with description "/home/bruno/tools/mrpt/libs/maps/src/maps/COccupancyGridMap2D_common.cpp:118: [mrpt::maps::COccupancyGridMap2D::COccupancyGridMap2D(float, float, float, float, float)] Called from here
```

Not sure what I am doing wrong.

Regards,
Bruno

closed 2020-09-17 by brunoeducsantos

jlblancoc, 2020-09-17:
BTW: I recommend you any of these forms:

```
mrpt::maps::COccupancyGridMap2D::Ptr map = mrpt::maps::COccupancyGridMap2D::Create(...);
auto map = mrpt::maps::COccupancyGridMap2D::Create(...);
auto map = std::make_shared<mrpt::maps::COccupancyGridMap2D>(...);
// (all three 100% equivalent)
```

to avoid the explicit call to new(), which *could* theoretically leak, and also calls the allocator twice.

brunoeducsantos, 2020-09-18:
Thanks a lot for the suggestion @jlblancoc . It worked much better.