MRPT/mrpt #1094: Unit test using COccupancy grid

brunoeducsantos, 2020-09-17:
Hi @jlblancoc ,
I am testing  a data structure conversion to _COccupancyGrid_ as follows:

```
TEST(GridMapAlignerTest, convertGridMapMRPT)
{
     //create global map
    GridMap global_map;
    float resolution = 0.1;
    grid_map::Length length = grid_map::Length(20., 20.);
    grid_map::Position position = grid_map::Position(5., 5.);
    global_map.setGeometry(length, resolution, position);
    global_map.add("test", 0.5);
    global_map.setFrameId("test");
    global_map.setBasicLayers({"test"});
    std::string path= "/home/bruno/grid_ws/src/skimap_path_planning/test/data/gridmatch_example.ini";
    GridMapAligner grid_aling(path);
    COccupancyGridMap2D::COccupancyGridMap2D::Ptr map(new COccupancyGridMap2D());
    const COccupancyGridMap2D content=*grid_aling.convertGridMapMRPT(global_map, "test"); 
    map->copyMapContentFrom(content);
    for (auto &val: map->getRawMap())
		{
       
            EXPECT_NEAR(val,0.606531,0.01);
        }
}
```
and 
```
mrpt::maps::COccupancyGridMap2D::Ptr GridMapAligner::convertGridMapMRPT(const grid_map::GridMap &grid_map, const std::string &layer)
{
	Position left_corner = grid_map.getPosition() - 0.5 * grid_map.getLength().matrix();
	Position upper_corner = grid_map.getPosition() + 0.5 * grid_map.getLength().matrix();
	COccupancyGridMap2D::Ptr map(new COccupancyGridMap2D(left_corner(0), left_corner(1), upper_corner(0), upper_corner(1), grid_map.getResolution()));
	for (GridMapIterator iterator(grid_map); !iterator.isPastEnd(); ++iterator)
	{
		
		Position robot_position;
		grid_map.getPosition(*iterator, robot_position);
		map->setPos(robot_position(0), robot_position(1), grid_map.at(layer, *iterator));
	}
	return map;
}
```

I am getting the following error:

```
unknown file: Failure
C++ exception with description "/home/bruno/tools/mrpt/libs/maps/src/maps/COccupancyGridMap2D_common.cpp:118: [mrpt::maps::COccupancyGridMap2D::COccupancyGridMap2D(float, float, float, float, float)] Called from here.
" thrown in the test body.
```
Any help would be appreciated.

Thanks,
Bruno

jlblancoc, 2020-09-17:
Hmm... wrap your test code with this to see the full exception stack: 

```
TEST(XXX)
{
try
{

} catch (const std::exception &e)
 {
  std::cerr << mrpt::exception_to_str(e);
 }
}
```

jlblancoc, 2020-09-17:
BTW: if this is a conversion from a ROS gridmap, you already have a converter [here](https://github.com/MRPT/mrpt/blob/develop/libs/ros1bridge/src/map.cpp#L84).

brunoeducsantos, 2020-09-17:
The issue was parameter order of initializing COccupancyGridMap. It wasn't according to class constructor.

closed 2020-09-17 by brunoeducsantos