MRPT/mrpt #1095: Conversion from COccupancyGridMap to CMultiMetricMap

brunoeducsantos, 2020-09-18:
Hi @jlblancoc ,

I am trying to convert `COccupancyGrid` to `CMultiMetricMap` using the following code:
```
auto occmap_localmap= std::make_shared<mrpt::maps::COccupancyGridMap2D>();
const mrpt::maps::CMultiMetricMap* multimap1 = nullptr;
multimap1= dynamic_cast<const mrpt::maps::CMultiMetricMap*>(occmap_localmap.get());
```
After that I am using it on grid matching using `Align` method as follows:

```
auto pose= gridAlign.Align(multimap1, multimap2, mrpt::poses::CPose2D(0,0,0), info);
```
The following error is coming up:

```
Thread 1 "local_mapping-t" received signal SIGSEGV, Segmentation fault.
0x00007ffff5df111a in mrpt::slam::CGridMapAligner::AlignPDF_robustMatch(mrpt::maps::CMetricMap const*, mrpt::maps::CMetricMap const*, mrpt::poses::CPosePDFGaussian const&, mrpt::slam::CGridMapAligner::TReturnInfo&) () from /usr/local/lib/libmrpt-slam.so.2.1
(gdb) bt
#0  0x00007ffff5df111a in mrpt::slam::CGridMapAligner::AlignPDF_robustMatch(mrpt::maps::CMetricMap const*, mrpt::maps::CMetricMap const*, mrpt::poses::CPosePDFGaussian const&, mrpt::slam::CGridMapAligner::TReturnInfo&) () from /usr/local/lib/libmrpt-slam.so.2.1
#1  0x00007ffff5df69b3 in mrpt::slam::CGridMapAligner::AlignPDF(mrpt::maps::CMetricMap const*, mrpt::maps::CMetricMap const*, mrpt::poses::CPosePDFGaussian const&, std::optional<std::reference_wrapper<mrpt::slam::TMetricMapAlignmentResult> >) () from /usr/local/lib/libmrpt-slam.so.2.1
#2  0x00007ffff5e40acb in mrpt::slam::CMetricMapsAlignmentAlgorithm::Align(mrpt::maps::CMetricMap const*, mrpt::maps::CMetricMap const*, mrpt::poses::CPose2D const&, std::optional<std::reference_wrapper<mrpt::slam::TMetricMapAlignmentResult> >) () from /usr/local/lib/libmrpt-slam.so.2.1
#3  0x00007ffff77d0e38 in GridMapAligner::Align(mrpt::maps::COccupancyGridMap2D const*, mrpt::maps::COccupancyGridMap2D const*) () from /home/bruno/grid_ws/devel/lib/liblocal_mapping.so
```
Thanks for your help,
Bruno

jlblancoc, 2020-09-18:
Hi, 

Check out the docs for [CMultiMetricMap](https://docs.mrpt.org/reference/latest/class_mrpt_maps_CMultiMetricMap.html) (unfortunately, recently all docs URLs changed and I'm aware googling for classes don't work as well as it used to...). 
It's a container of maps, so you don't have to convert between types, but to insert the grid into the container: 

```cpp
auto occmap_localmap= std::make_shared<mrpt::maps::COccupancyGridMap2D>();

mrpt::maps::CMultiMetricMap m;
m.maps.push_back(occmap_localmap);
```

brunoeducsantos, 2020-09-18:
Thanks for you feedback.
I did something like this:
```
mrpt::poses::CPosePDFSOG::Ptr GridMapAligner::AlignPDF(mrpt::maps::COccupancyGridMap2D::Ptr map1, const mrpt::maps::COccupancyGridMap2D::Ptr map2, const mrpt::poses::CPosePDFGaussian &initialEstimationPDF)
{	

	mrpt::poses::CPose2D estimateMean;
	mrpt::math::CMatrixDouble33 estimateCOV;
	mrpt::maps::CMultiMetricMap multimap1;
	mrpt::maps::CMultiMetricMap multimap2;
	multimap1.maps.push_back(map1);
	multimap2.maps.push_back(map2);
	mrpt::poses::CPosePDF::Ptr parts = gridAlign.AlignPDF(&multimap1, &multimap2, initialEstimationPDF, info);
	mrpt::poses::CPosePDFSOG::Ptr pdf_SOG = std::dynamic_pointer_cast<mrpt::poses::CPosePDFSOG>(parts);
	pdf_SOG->getMostLikelyCovarianceAndMean(estimateCOV, estimateMean);
	return pdf_SOG;
}
```
Got the following error:

```
 error: no matching function for call to ‘std::deque<mrpt::containers::deepcopy_poly_ptr<std::shared_ptr<mrpt::maps::CMetricMap> > >::push_back(const Ptr&)’
```

brunoeducsantos, 2020-09-18:
I follow your advice as :+1: 
```
mrpt::poses::CPosePDFSOG::Ptr GridMapAligner::AlignPDF(mrpt::maps::COccupancyGridMap2D::Ptr map1,  mrpt::maps::COccupancyGridMap2D::Ptr map2, const mrpt::poses::CPosePDFGaussian &initialEstimationPDF)
{	

	mrpt::poses::CPose2D estimateMean;
	mrpt::math::CMatrixDouble33 estimateCOV;
	mrpt::maps::CMultiMetricMap multimap1;
	mrpt::maps::CMultiMetricMap multimap2;
	multimap1.maps.push_back(map1);
	multimap2.maps.push_back(map2);
	mrpt::poses::CPosePDF::Ptr parts = gridAlign.AlignPDF(&multimap1, &multimap2, initialEstimationPDF, info);
	mrpt::poses::CPosePDFSOG::Ptr pdf_SOG = std::dynamic_pointer_cast<mrpt::poses::CPosePDFSOG>(parts);
	pdf_SOG->getMostLikelyCovarianceAndMean(estimateCOV, estimateMean);
	return pdf_SOG;
}
```
And got the following error:
```
: error: no matching function for call to ‘std::deque<mrpt::containers::deepcopy_poly_ptr<std::shared_ptr<mrpt::maps::CMetricMap> > >::push_back(std::__shared_ptr_access<mrpt::maps::COccupancyGridMap2D, (__gnu_cxx::_Lock_policy)2, false, false>::element_type&)’
```

brunoeducsantos, 2020-09-21:
```
emplace_back 
```
sort the problem

closed 2020-09-21 by brunoeducsantos