MRPT/mrpt #1096: CGridAligner/Align function taking too much on threads

brunoeducsantos, 2020-09-23:
Hi @jlblancoc ,
I am using `CGridALigner` function `Align` and it shows the following as expected:

> [WARNING][selectGoodFeaturesKLT]: Only 247 of 805 points could be extracted in the image.
> 
> [WARNING][selectGoodFeaturesKLT]: Only 93 of 805 points could be extracted in the image.
> [13:37:17.2863|INFO |CMetricMapsAlignmentAlgorithm] [CGridMapAligner] Overall estimation(247 corrs, total: 22971): (108.162,-15.365,77.792deg)
> [13:37:17.3815|INFO |CMetricMapsAlignmentAlgorithm] SOG mode: (147.338,-0.120,89.90deg) inliers: 27
> [13:37:17.3816|INFO |CMetricMapsAlignmentAlgorithm] SOG mode: (147.428,-0.011,89.99deg) inliers: 63
> [13:37:17.3816|INFO |CMetricMapsAlignmentAlgorithm] SOG mode: (147.485,0.026,90.02deg) inliers: 63
> [13:37:17.3816|INFO |CMetricMapsAlignmentAlgorithm] SOG mode: (147.798,0.172,90.12deg) inliers: 34
> [13:37:17.3816|INFO |CMetricMapsAlignmentAlgorithm] SOG mode: (147.423,-0.036,89.98deg) inliers: 63
> [13:37:17.3817|INFO |CMetricMapsAlignmentAlgorithm] SOG mode: (147.798,0.163,90.12deg) inliers: 34
> [13:37:17.3817|INFO |CMetricMapsAlignmentAlgorithm] SOG mode: (147.504,-0.013,90.01deg) inliers: 62
> [13:37:17.3817|INFO |CMetricMapsAlignmentAlgorithm] SOG mode: (147.463,-0.001,90.00deg) inliers: 65
> [13:37:17.3818|INFO |CMetricMapsAlignmentAlgorithm] [CGridMapAligner] amModifiedRANSAC: 8 SOG modes merged to 2 in 0.000sec
> [13:37:17.3819|INFO |CMetricMapsAlignmentAlgorithm] Largest consensus: 65
> 

Although, after this log it takes a few mins until the threads are killed. Is it normal to take so long ?

Thanks,
Bruno

brunoeducsantos, 2020-09-23:
Flag for saving pairs was activated.

closed 2020-09-23 by brunoeducsantos

jlblancoc, 2020-09-23:
> Flag for saving pairs was activated.

If you think it would more practical to have it disabled by default, feel free of opening a PR!