MRPT/mrpt #1104: Cannot get example working on Windows

Sirver51, 2020-10-11:
I don't know if I'm doing something wrong or I just have terrible luck, but I've been trying for more than a week now to get a build working on Windows 10 and I'm at my wit's end. I haven't been keeping step-by-step logs of what I've been doing, because I didn't think I'd have to open an issue for this, so some parts of the following may be fuzzy, for which I apologize; I hope it's enough information that someone will be able to point me in the right direction. Here's what I did (I've attached [the test program, CMakeLists.txt, and sample data](https://github.com/MRPT/mrpt/files/5360423/test.zip) that I've been using):

I've tried using binaries as well as building from source with every possible toolchain: VS2017 x64, VS2019 x64, MinGW, and Nmake, the latter two of which would not build successfully. Builds with Visual Studio succeeded for the most part, with 19 of the tests failing, but the tools like SceneViewer3D and RawLogViewer worked fine, so I assume it's okay; I didn't make a note of which ones failed, since I don't think it's relevant, but if I'm wrong I will rebuild and post that if necessary.

With both the pre-built binaries and the self-built versions (currently on pre-built), after a bunch of CMake and path wrangling, the test program errors out with the same exceptions in the debugger:

>`Exception thrown at 0x00007FFE469A2B9D (ucrtbase.dll) in slam_icp_simple_example.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.`
>`Exception thrown at 0x00007FFE466D3E49 in slam_icp_simple_example.exe: Microsoft C++ exception: std::_With_nested<std::logic_error,std::logic_error> at memory location 0x000000E27CB2D3B0.`

And running it from CMD gives this:

> `MRPT exception caught: C:\projects\mrpt\libs\maps\src\maps\CPointsMap.cpp:160: [mrpt::maps::CPointsMap::load2Dor3D_from_text_file] Called from here.`

Which is confusing, because I don't have a directory called `C:\projects`; my install is in `C:\mrpt-2.1.0`, as put there by the installer.

I also tried building and running the "slam_icp_simple_example" example (the one I cannibalized to make the test program) unchanged, and got this:

> `Exception thrown at 0x00007FFE2EA016E9 (vcruntime140.dll) in slam_icp_simple_example.exe: 0xC0000005: Access violation writing location 0x0000000000000000.`
> `Unhandled exception at 0x00007FFE2EA016E9 (vcruntime140.dll) in slam_icp_simple_example.exe: 0xC0000005: Access violation writing location 0x0000000000000000.`

I've tried running all this on an Ubuntu 18.04 VM, and it works just fine there.

If anyone has any input at all, it would be greatly appreciated.

#### P.S.: CMake
About halfway through while using the prebuilt binaries, providing my own version of Eigen caused CMake to give this error, despite it having worked fine before.:

>`CMake Error at C:/mrpt-2.1.0/cmake/EP_eigen3-config.cmake:5 (add_library): add_library cannot create target "Eigen" because another target with the same name already exists.  The existing target is an interface library created in source directory "D:/Code/asv/mrpt/test".  See documentation for policy CMP0002 for more details.`

I don't quite recall if it was before or after I started providing my own install of glfw3, but the long and the short of it is that even undoing whatever I did didn't fix it, and I had to resort to setting the CMP0002 policy to OLD to get it to work. The final exception was the same before and after this started happening, so I don't think it's relevant, but I'm mentioning it anyway just in case.

jlblancoc, 2020-10-15:
Hi, 
I'm sorry for the bad experience! Binaries for Windows are not as well tested as Linux packages, and many potential problems regarding different library and compiler versions, etc. arise. 

But I'm surprised you have problems with your own compiled versions... Try compiling and running in Debug to check where exactly happens that invalid memory access. It might be a typical mix-of-dlls-problem, or a real bug in mrpt for Windows.

Sirver51, 2020-10-16:
OK, I'm not sure what happened, but my test program is now working; a basic test of the GUI and OpenGL modules seems to work as well, whereas they didn't before. I didn't change any of the code, I'm still using the pre-built binaries, and as far as I'm aware nothing about my system has changed either, except maybe a Windows update; I'm genuinely clueless as to how this could have happened.

@jlblancoc, about your suggestion, I did try running in Debug, but once I hit `ucrtbase.dll`, the debugger couldn't help, since that one's a Windows DLL and doesn't have debug symbols or source files or anything. All I know is that it originated from `mrpt::maps::CPointsMap::load2Dor3D_from_text_file`. And that it's not causing a problem anymore, for some reason.

Any idea as to what might have caused the miraculous recovery? 😅

By the way, while my test program is working, the `slam_icp_simple_example` is not - I still get the same error about `vcruntime140.dll` I mentioned in the original comment. It originated from:

`CObservation2DRangeScan scan1;`
`stock_observations::example2DRangeScan(scan1, 0);`

On the second line. So maybe an issue with the stock_observations module? I'll try test it with some external data in the morning and see if that works.

jlblancoc, 2021-02-15:
It looks like a mixture of compiler versions: if the user compiler (in Windows, MSVC) is not exactly the one used to generate the binaries, they may not be ABI-compatible and building from sources may be required.

I'll close due to lack of follow-up.

closed 2021-02-15 by jlblancoc