MRPT/mrpt #1113: Convert PointCloud2 message to

brunoeducsantos, 2020-10-23:
Hi @jlblancoc ,

I am trying to convert `PointCloud2` message to `CObservationRotatingScan` type. For some reason, it isn't converting. 
Here it is the snippet:

```
	auto dense_occmap = std::make_shared<mrpt::maps::COccupancyGridMap2D>();
	auto observation_scan = std::make_shared<mrpt::obs::CObservationRotatingScan>();
	auto sensor_pose = std::make_shared<mrpt::poses::CPose3D>();
	Eigen::Matrix4d t_map_odom;
	t_map_odom = transformFrames("map", "odom", 0.3);
	auto euler = t_map_odom.topLeftCorner<3, 3>().eulerAngles(0, 1, 2);
	auto position_robot = t_map_odom.block<3, 1>(0, 3);
	sensor_pose->setFromValues(position_robot(0), position_robot(1), position_robot(2), euler(2), euler(0), euler(1));
	if (mrpt::ros1bridge::fromROS(*cloud, *observation_scan, *sensor_pose))
	{
		dense_occmap->insertObservation(*observation_scan);
		ROS_INFO_STREAM("SUCCESS");
	}
```

Thanks,
Bruno

jlblancoc, 2020-10-23:
I'll check it.

But anyway, it would be far more efficient to directly be able to acquire the sensor data as a matrix as used inside the rotating lidar observation class...

If you don't mind, write me directly to explore some options to do this: JLBLANCO@UAL.ES

brunoeducsantos, 2020-10-24:
Thanks a lot Jose. I will send you an email asap. 
Thanks for your help.

jlblancoc, 2020-10-27:
Many simplifications can be done to the code above. First, unless you need shared_ptr's for some good reason, prefer storing things on the stack  (faster): 

``` 
mrpt::maps::COccupancyGridMap2D dense_occmap;
mrpt::obs::CObservationRotatingScan observation_scan;
const Eigen::Matrix4d t_map_odom = transformFrames("map", "odom", 0.3);
auto sensor_pose = mrpt::poses::CPose3D(mrpt::math::CMatrixDouble44(t_map_odom));
if (mrpt::ros1bridge::fromROS(cloud, observation_scan, sensor_pose))
{
	dense_occmap.insertObservation(observation_scan);
	ROS_INFO_STREAM("SUCCESS");
}
```

Give it a try, I didn't compile it but it should be ok, typos apart. 

Now,  the conversion into `observation_scan` should work, but the implementation of inserting it into the occupancy grid is not for the type CObservationRotatingScan. I'll answer more details on the email.

brunoeducsantos, 2020-10-29:
Sorry the late answer. 
Thanks a lot for your feedback. 
I will close the issue since my question about conversion is solved.

closed 2020-10-29 by brunoeducsantos