MRPT/mrpt #1116: Errors On OSX

Gatsby23, 2020-11-29:


Gatsby23, 2020-11-29:
When I compile mrpt on my macbook pro, it shows there are something wrong with CschemeArchive.h file, like this:
[image]

jlblancoc, 2020-11-30:
Hi, 
What's your system setup? 
- OSX / compiler version
- version of "json", which seems to be installed manually under /usr/local.

Gatsby23, 2020-11-30:
> Hi,
> What's your system setup?
> 
> * OSX / compiler version
> * version of "json", which seems to be installed manually under /usr/local.

[image]
It's the default version of Macos Big Sur, version 11.01
I don't remember what kind of  json  I have installed or not ? How could I check it ?

jlblancoc, 2020-12-01:
I can't directly test builds on a Mac, but here we have CI with an OSX pipeline and it's been building ok for a long time, so there must be something different (a newer/older compiler or json library?). 
Please, run "cmake ." at your build directory and show the "paragraph" after "PLATFORM". For reference, this OSC setup works in CI (look for the "PLATFORM" summary with all the component versions):  https://app.circleci.com/pipelines/github/MRPT/mrpt/1867/workflows/83a12435-5895-4004-a67a-54ca06571cf7/jobs/11254/parallel-runs/0/steps/0-109

Gatsby23, 2020-12-01:
> I can't directly test builds on a Mac, but here we have CI with an OSX pipeline and it's been building ok for a long time, so there must be something different (a newer/older compiler or json [...]
OK, here is my configure list
[image]
[image]
[image]

jlblancoc, 2020-12-05:
This was actually caused by #1118, which will be closed by PR #1121 briefly.

closed 2020-12-05 by jlblancoc