MRPT/mrpt #1118: mrpt: build failure with latest libjsoncpp

LocutusOfBorg, 2020-12-04:
Hello, looks like the new libjsoncpp is making mrpt FTBFS

```
[ 22%] Building CXX object libs/serialization/CMakeFiles/serialization.dir/src/CSchemeArchiveBase.cpp.o
cd /mrpt-2.1.4/obj-x86_64-linux-gnu/libs/serialization && /usr/bin/c++ -Dserialization_EXPORTS -I/usr/include/suitesparse -I/mrpt-2.1.4/libs/serialization/src -I/mrpt-2.1.4/libs/serialization/include -I/mrpt-2.1.4/libs/rtti/include -I/mrpt-2.1.4/libs/core/include -I/mrpt-2.1.4/obj-x86_64-linux-gnu/include/mrpt-configuration -I/mrpt-2.1.4/libs/typemeta/include -isystem /usr/include/jsoncpp -g -O2 -fdebug-prefix-map=/mrpt-2.1.4=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -O3 -DNDEBUG -fPIC -Wall -Wabi=11 -Wno-long-long -Wno-variadic-macros -Wshadow -Wreturn-local-addr -Werror=return-local-addr -Wno-ignored-attributes -Wno-int-in-bool-context -Wno-write-strings -Wreturn-type -Werror=return-type -Wformat -Werror=format-security -Wextra -Wtype-limits -Wcast-align -Wparentheses -Wno-unused-parameter -mtune=native -O3 -msse2 -fPIC -std=gnu++17 -o CMakeFiles/serialization.dir/src/CSchemeArchiveBase.cpp.o -c /mrpt-2.1.4/libs/serialization/src/CSchemeArchiveBase.cpp
In file included from /mrpt-2.1.4/libs/serialization/src/CSchemeArchive.cpp:13:
/mrpt-2.1.4/libs/serialization/include/mrpt/serialization/CSchemeArchive.h: In instantiation of 'mrpt::serialization::CSchemeArchiveBase& mrpt::serialization::CSchemeArchive<SCHEME_CAPABLE>::operator=(std::nullptr_t) [with SCHEME_CAPABLE = Json::Value; std::nullptr_t = std::nullptr_t]':
/mrpt-2.1.4/libs/serialization/include/mrpt/serialization/CSchemeArchive.h:73:43:   required from here
/mrpt-2.1.4/libs/serialization/include/mrpt/serialization/CSchemeArchive.h:76:9: error: use of deleted function 'Json::Value::Value(std::nullptr_t)'
   76 |   m_val = val;
      |   ~~~~~~^~~~~
In file included from /usr/include/jsoncpp/json/reader.h:11,
                 from /usr/include/jsoncpp/json/json.h:11,
                 from /mrpt-2.1.4/libs/serialization/src/CSchemeArchive.cpp:18:
/usr/include/jsoncpp/json/value.h:345:3: note: declared here
  345 |   Value(std::nullptr_t ptr) = delete;
      |   ^~~~~
make[4]: *** [libs/serialization/CMakeFiles/serialization.dir/build.make:111: libs/serialization/CMakeFiles/serialization.dir/src/CSchemeArchive.cpp.o] Error 1
make[4]: *** Waiting for unfinished jobs....

```
Can you please have a look?
thanks

jlblancoc, 2020-12-05:
Awesome, that explains #1116 .
I'll take a look now that we know where's the actual problem.

closed 2020-12-06 by jlblancoc

LocutusOfBorg, 2020-12-07:
btw I saw some package on mentors.debian.org, but it has disappeared... do you need somebody to upload the fix?

jlblancoc, 2020-12-07:
Thanks! It was a first test, but *now* it's already submitted to mentors again. If you want to check and upload it, it would be fantastic :+1:
I just finished uploading it, it will appear in mentors within minutes...

LocutusOfBorg, 2020-12-07:
uploaded thanks!

PR #1121 (2020-12-06): Fix build error against libjsoncpp 1.9.4
## Changed apps/libraries

* mrpt-serialization

## PR Description

Remove possibility of assigning a scheme value from a `nullptr`. It was removed in libjsoncpp.

Perhaps it might be needed to add an alternative method to set a given json/yaml/... node as "null", but apparently no code was using this feature, so it was just removed. Add if needed in the future.

Closes #1118