MRPT/mrpt #1123: mrpt_bfd_error_handler_type in linux 16.04

songming1995, 2021-01-10:
/home/songming/Downloads/mrpt-2.1.7/libs/core/src/backtrace.cpp: In function ‘std::vector<mrpt::TCallStackEntry> backtraceSymbols(void* const*, int)’:
/home/songming/Downloads/mrpt-2.1.7/libs/core/src/backtrace.cpp:271:24: error: invalid conversion from ‘void (*)(const char*, __va_list_tag*)’ to ‘bfd_error_handler_type {aka void (*)(const char*, ...)}’ [-fpermissive]
  bfd_set_error_handler(&mrpt_bfd_error_handler_type);
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/songming/Downloads/mrpt-2.1.7/libs/core/src/backtrace.cpp:44:0:
/usr/include/bfd.h:6856:24: note:   initializing argument 1 of ‘void (* bfd_set_error_handler(bfd_error_handler_type))(const char*, ...)’
 bfd_error_handler_type bfd_set_error_handler (bfd_error_handler_type);
                        ^~~~~~~~~~~~~~~~~~~~~
libs/core/CMakeFiles/core.dir/build.make:134: recipe for target 'libs/core/CMakeFiles/core.dir/src/backtrace.cpp.o' failed
make[2]: *** [libs/core/CMakeFiles/core.dir/src/backtrace.cpp.o] Error 1
CMakeFiles/Makefile2:2110: recipe for target 'libs/core/CMakeFiles/core.dir/all' failed
make[1]: *** [libs/core/CMakeFiles/core.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

songming1995, 2021-01-10:
#---------------------------------------------
# Macro declared in "DeclareMRPTLib.cmake":
#---------------------------------------------
define_mrpt_lib(
	# Lib name
	core
	# Dependencies
	)

if(BUILD_mrpt-core)
	# Includes: <mrpt/config.h> & <mrpt/version.h> config headers:
	target_include_directories(core PUBLIC
		$<BUILD_INTERFACE:${MRPT_CONFIG_FILE_INCLUDE_DIR}>
		$<INSTALL_INTERFACE:include>
	)

	target_link_libraries(core PRIVATE Threads::Threads) # for WorkerThreads

	# Enforce C++17 in all dependent projects:
	mrpt_lib_target_requires_cpp17(core)

	# Fix a MSVC binary-breaking compatibility in MSVC 2017 15.8:
	if (MSVC)
		target_compile_definitions(core PUBLIC -D_ENABLE_EXTENDED_ALIGNED_STORAGE)
	endif()

	if(WIN32)
		target_link_libraries(core PRIVATE Dbghelp) # For mrpt::demangle(), mrpt::callStackBackTrace()
	else()
		target_link_libraries(core PRIVATE ${CMAKE_DL_LIBS}) # For mrpt::callStackBackTrace()
	endif()

	if (CMAKE_MRPT_HAS_BFD)
		target_link_libraries(core PRIVATE ${BFD_LIBRARIES})
	endif()
	
	if (MRPT_EXCEPTIONS_WITH_CALL_STACK)
		target_compile_definitions(core PRIVATE MRPT_EXCEPTIONS_WITH_CALL_STACK)
	endif()

endif()

songming1995, 2021-01-10:
define_mrpt_lib(
	# Lib name
	core
	# Dependencies
	)

if(BUILD_mrpt-core)
	# Includes: <mrpt/config.h> & <mrpt/version.h> config headers:
	target_include_directories(core PUBLIC
		$<BUILD_INTERFACE:${MRPT_CONFIG_FILE_INCLUDE_DIR}>
		$<INSTALL_INTERFACE:include>
	)

	target_link_libraries(core PRIVATE Threads::Threads) # for WorkerThreads

	# Enforce C++17 in all dependent projects:
	mrpt_lib_target_requires_cpp17(core)

	# Fix a MSVC binary-breaking compatibility in MSVC 2017 15.8:
	if (MSVC)
		target_compile_definitions(core PUBLIC -D_ENABLE_EXTENDED_ALIGNED_STORAGE)
	endif()

	if(WIN32)
		target_link_libraries(core PRIVATE Dbghelp) # For mrpt::demangle(), mrpt::callStackBackTrace()
	else()
		target_link_libraries(core PRIVATE ${CMAKE_DL_LIBS}) # For mrpt::callStackBackTrace()
	endif()

	if (CMAKE_MRPT_HAS_BFD)
		target_link_libraries(core PRIVATE ${BFD_LIBRARIES})
	endif()
	
	if (MRPT_EXCEPTIONS_WITH_CALL_STACK)
		target_compile_definitions(core PRIVATE MRPT_EXCEPTIONS_WITH_CALL_STACK)
	endif()

endif()

songming1995, 2021-01-10:
# Includes: <mrpt/config.h> & <mrpt/version.h> config headers:
	target_include_directories(core PUBLIC
		$<BUILD_INTERFACE:${MRPT_CONFIG_FILE_INCLUDE_DIR}>
		$<INSTALL_INTERFACE:include>
	)

	target_link_libraries(core PRIVATE Threads::Threads) # for WorkerThreads

	# Enforce C++17 in all dependent projects:
	mrpt_lib_target_requires_cpp17(core)

	# Fix a MSVC binary-breaking compatibility in MSVC 2017 15.8:
	if (MSVC)
		target_compile_definitions(core PUBLIC -D_ENABLE_EXTENDED_ALIGNED_STORAGE)
	endif()

	if(WIN32)
		target_link_libraries(core PRIVATE Dbghelp) # For mrpt::demangle(), mrpt::callStackBackTrace()
	else()
		target_link_libraries(core PRIVATE ${CMAKE_DL_LIBS}) # For mrpt::callStackBackTrace()
	endif()

	if (CMAKE_MRPT_HAS_BFD)
		target_link_libraries(core PRIVATE ${BFD_LIBRARIES})
	endif()
	
	if (MRPT_EXCEPTIONS_WITH_CALL_STACK)
		target_compile_definitions(core PRIVATE MRPT_EXCEPTIONS_WITH_CALL_STACK)
	endif()

songming1995, 2021-01-10:
mrpt uses private binutils shared libraries:

Package: libmrpt-core2.1
Depends: libbinutils (>= 2.35.1), libbinutils (<< 2.35.2), [...]

jlblancoc, 2021-02-15:
Yes, `libiberty-dev` is a new *optional* dependency to enable the feature of fully demangled stack traces with source code file names and line numbers. 

Please, reopen if this is still an issue.

closed 2021-02-15 by jlblancoc

ketchart, 2021-09-02:
https://ibb.co/wgzLtGY

I tried sudo apt-get install libiberty-dev, it did not work
Please guide me

jlblancoc, 2021-09-05:
Hi @ketchart ! 
Fixes for u16.04 are not supported anymore, *but* if that's your unique problem, just set `DISABLE_BFD=OFF` in CMake (using `cmake-gui` or `ccmake`) and you should get rid of the problem. 
Best,

ketchart, 2021-09-07:
https://ibb.co/Sdw9cw1

I did like that (then I click OK -> Configure -> Generate).  Error appeared again when I did "make -j8",  I believe I did it wrong
Please recorrect me

jlblancoc, 2021-09-07:
Ah! I forgot to mention that the existing cmake option is "advanced", so
it's hidden by default: first select "show advanced" in cmake-gui, then
check that variable, configure, generate, and build...

ketchart, 2021-09-09:
https://ibb.co/txw17SX

"mrpt_bfd_error_handler_type" error is solved. But, a new error appeared during "cmake --build ." 

https://ibb.co/sm3WLrF

Actually, I should Post it as a new Issue. 
Please advise me for the further step.

jlblancoc, 2021-09-14:
@ketchart  Yes, please, open a new issue ticket, and attach the text file `CMAKE_CONFIG.txt` with the result of: 

    cd mrpt
    mkdir build && cd build
    VERBOSE=1 cmake ..  | tee CMAKE_CONFIG.txt