MRPT/mrpt #1136: Memory leak or not the right way of using CImage with opengl::CTexturedPlane

nrasulnrasul, 2021-02-16:
Hi,

I am using `gui::CDisplayWindow3D` and `opengl::CTexturedPlane `for visualizing maps(colour maps not just gray).  
I observed that there was memory leak in `opengl::CTexturedPlane::assignImage `.   
If the below program run for 1000 loops (x =1000), it will fill the 16gb ram and the system hangs.

I have attached the program with cmake, so that you can build and verify the issue easily. Just run the program and observe the memory using `htop`.

[ZIP file](https://github.com/MRPT/mrpt/files/5988353/cimage_leak.zip)

Is it the correct way of using `CImage` with `CTexturedPlane`. 

Thanks.

```
#include <iostream>
#include <mrpt/img/CImage.h>
#include <mrpt/opengl/CTexturedPlane.h>
#include <mrpt/gui/CDisplayWindow3D.h>

using namespace mrpt;
using namespace mrpt::img;
using namespace mrpt::gui;
using namespace mrpt::opengl;
int main()
{
long x = 0;

CDisplayWindow3D *window3D = new CDisplayWindow3D();
window3D->setWindowTitle("Visualization");
window3D->resize(800, 600);

opengl::CTexturedPlane::Ptr objs = opengl::CTexturedPlane::Create();
objs->setName("map");
objs->setPlaneCorners(0, 0, 0, 0);

opengl::COpenGLScene::Ptr &ptrScene = window3D->get3DSceneAndLock();
ptrScene->insert(objs);
window3D->unlockAccess3DScene();
window3D->forceRepaint();

while (x< 500)
{
	
	//mrpt::img::CImage::Ptr imgColor = std::make_shared<mrpt::img::CImage>(1600, 1600, mrpt::img::TImageChannels::CH_RGB);
	//mrpt::img::CImage::Ptr imgTrans = std::make_shared<mrpt::img::CImage>(1600, 1600, mrpt::img::TImageChannels::CH_GRAY);

	mrpt::img::CImage imgColor(1600, 1600, mrpt::img::TImageChannels::CH_RGB);	
	mrpt::img::CImage imgTrans(1600, 1600, mrpt::img::TImageChannels::CH_GRAY);

	for (uint32_t j = 0; j < 1600; j++) {
			for (uint32_t i = 0; i < 1600; i++) {

			//imgTrans->setPixel(i, j, img::TColor::red());
			//imgColor->setPixel(i, j, img::TColor::green());
			imgColor.setPixel(i, j, img::TColor::red());
			imgTrans.setPixel(i, j, img::TColor::red());
		}
	}

	{
		opengl::COpenGLScene::Ptr &ptrScene = window3D->get3DSceneAndLock();
		opengl::CTexturedPlane::Ptr ptrPlane = std::dynamic_pointer_cast<opengl::CTexturedPlane>(ptrScene->getByName("map"));
		ptrPlane->setPlaneCorners(0, 0, 0, 0);
		ptrPlane->setPlaneCorners(0, 0, 1599, 1599);

		
		//ptrPlane->assignImage(*imgColor, *imgTrans);
		ptrPlane->assignImage(imgColor, imgTrans); // causing memory leak
		ptrPlane->notifyChange();
		window3D->unlockAccess3DScene();
		window3D->forceRepaint();
	}

	std::cout << "x = " << ++x <<std::endl;
}

if(window3D) delete window3D;
return 0;
}
```

jlblancoc, 2021-02-16:
Thanks! Confirmed, I can see the memory leak. I'll look at it...

jlblancoc, 2021-02-17:
Fixed. See: a4b8c37b81de7340d5f334f0d9eb67be5244a8c1  and the updated source code:
[main.zip](https://github.com/MRPT/mrpt/files/5993099/main.zip)

Cheers

nrasulnrasul, 2021-02-17:
Thanks for the fix and updated code. I couldnt verify it now becuase i am using mrpt from [ppa:joseluisblancoc/mrpt](ppa:joseluisblancoc/mrpt) repository in Ubuntu 20.04.

jlblancoc, 2021-02-17:
The most relevant wrong part in your code was the wrong order of xMin, xMax
setting the plane dimensions, you can change that locally.

nrasulnrasul, 2021-02-17:
I have tested your code, but I could still see the memory leak with looping 1000 times. 
`i.e  while (x < 1000) {   // Line 48`

jlblancoc, 2021-02-18:
Ok, I see... you were right, thanks a lot for testing and reporting!
This serious leak is now fixed in the branch `develop`.

commit a4b8c37b (2021-02-17): setPlaneCorners() should check for incorrect values. (Closes #1136)
commit 75fc0429 (2021-02-18): Fix opengl texture memory leak

Closes #1136
closed 2021-02-18 by jlblancoc