MRPT/mrpt #1137: Seg. fault using camera calibration

Svato, 2021-02-17:
`(camera-calib:26510): Gtk-CRITICAL **: 18:31:52.023: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton

(camera-calib:26510): Gtk-CRITICAL **: 18:31:52.023: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkSpinButton
Segmentation fault (core dumped)
`

jlblancoc, 2021-02-18:
Can you provide more details? 
- When does the crash exactly happen? (Steps to reproduce)
- OS and MRPT version.

jlblancoc, 2021-03-08:
Closing for lack of feedback. Reopen if needed!

Svato, 2021-03-18:
LINUX Mint mate / Win 10. 
I did follow video - https://www.youtube.com/watch?v=BkZkq6zPwQM
I did make 20 pictures. After pressing calibrate i got the segmentation error

jlblancoc, 2021-03-18:
ok. Could you please share your 20 images in a zip as a comment attachment?

Likhith-Sugganahalli, 2021-04-15:
Hi, i have run into the same problem, after taking the 15 frames, clicking on calibration starts the process, in the log, it detects all images, first couple of roll pitch statements are printed, and it crashes

[image]

[image]

Where can i access the images from?

jlblancoc, 2021-04-15:
Thanks! Fantastic, now I could reproduce the bug: indeed the problem only happens after grabbing the images from a live camera, I formerly only tested it against images in the disk and in that case it doesn't fail. 

I'll fix it soon.

jlblancoc, 2021-04-15:
It's fixed now in `develop`, thanks for reporting!

commit c5db4446 (2021-04-15): BUGFIX: Fix crash in camera-calib

Closes #1137
closed 2021-04-15 by jlblancoc