MRPT/mrpt #1141: Infinite recursion in basic exception handler

jolting, 2021-02-27:
https://github.com/MRPT/mrpt/blob/6cf7cd291eccd9ec101442fd22b0629f71b3cda8/libs/core/src/exceptions.cpp#L95

This should be
```
const auto* ecb = dynamic_cast<const ExceptionWithCallBackBase*>(&e);
if(ecb)
{
  ret = ecb->originalWhat;
}
else
{
  ret = e->what();
}
```
Since m_what is empty.

call exception_to_str
https://github.com/MRPT/mrpt/blob/78c8f30facd9994880a2b16157e55e25b93b2312/libs/core/src/exceptions.cpp#L105
call  impl_excep_to_str
https://github.com/MRPT/mrpt/blob/78c8f30facd9994880a2b16157e55e25b93b2312/libs/core/src/exceptions.cpp#L51
call basic version
https://github.com/MRPT/mrpt/blob/78c8f30facd9994880a2b16157e55e25b93b2312/libs/core/src/exceptions.cpp#L100
call e.what(), m_what is still empty
https://github.com/MRPT/mrpt/blob/78c8f30facd9994880a2b16157e55e25b93b2312/libs/core/include/mrpt/core/exceptions.h#L95
repeat steps 1-4.

jolting, 2021-02-27:
Potential fix:
https://github.com/jolting/mrpt/tree/infiniteExceptions

jlblancoc, 2021-02-27:
Wow!! Excellent finding.

closed 2021-02-28 by jlblancoc