MRPT/mrpt #1148: Windows binaries not working - Entry point not found

X3msnake, 2021-03-07:
[image]

Any idea why this error?
I had to set path manually on windows envoirenement since the installer did not add them

jlblancoc, 2021-03-08:
Are you on a 32bit version of Windows? Published binaries are for 64bit only, it looks the easiest explanation at first sigh...

X3msnake, 2021-03-09:
negative. I am running win7 64bit

X3msnake, 2021-03-09:
should be something with the laptop or the os since i was able to run the
app on the desktop with win 10

No dia terça-feira, 9 de março de 2021, Vinicius Silva <x3msnake@gmail.com>

jlblancoc, 2022-03-21:
Closing, apparently not a real mrpt packages issue (?).

closed 2022-03-21 by jlblancoc