MRPT/mrpt #1150: can't compile with mingw64

Dharmesh946, 2021-03-12:
Hi

I tried to compile MRPT on windows with mingw64:
I have created a build directory next to the CMakeLists.txt file
```bash
cd build
cmake-gui ..
configure
generate
make
```

I have directly the following error:
Building CXX object libs/core/CMakeFiles/core.dir/src/WorkerThreadsPool.cpp.obj
In function 'void mySetThreadName(const string&, std::thread&)':
error: 'SetThreadDescription' was not declared in this scope
  SetThreadDescription(theThread.native_handle(), wName);

I can't find any reference to that issue anywhere.

Help would be appreciated. Regards

jlblancoc, 2021-03-12:
I'm curious: did you provide opencv or any other dependency? Using any
package manager or installer, or building from sources? It's been ages
since I last tried mingw64 :-)

I'll look at the build error, I can imagine the reason of this one.

If in the meanwhile you or someone else finds a patch to fix it, feel free
of opening a pull request !

Dharmesh946, 2021-03-15:
Hi,

I compile OpenCV and wxWidgets from sources (with mingw64) and provided them to MRPT.
For Eigen3 I'm currently using the embedded version though I've got  another one installed.
wxWidgets was compiled using instructions from MRPT web site.

Regards

jlblancoc, 2021-05-21:
Hi!

I fixed all mingw errors I could find. Could you give it a try now, please? (develop git branch)

Dharmesh946, 2021-06-04:
Hi,

Sorry for the delay.
I've just tried, same error:
In function 'void mySetThreadName(const string&, std::thread&)':
error: 'SetThreadDescription' was not declared in this scope
  SetThreadDescription(theThread.native_handle(), wName);
note: suggested alternative: 'SetThreadDesktop'
  SetThreadDescription(theThread.native_handle(), wName);
Regards

jlblancoc, 2021-06-04:
Hmm... have you updated your git repository first? Did you check out the `develop` branch? 🤨 😄

Dharmesh946, 2021-06-14:
Sorry, but yes I did...

D.

Dharmesh946, 2021-06-14:
FYI, cmake configuration:
OpenCV: custom install
wxwidgets: custom install following your directives
Build type: release
custom install directory
don't use QT
mingw generator

Dharmesh946, 2021-06-14:
Perhaps an idea: I just happen to realize that my default mingw 64 installation (from https://sourceforge.net/projects/mingw-w64/) does not properly support std::thread.
The standard headers are there so my test programs compiles but I've got a runtime error: missing entry point for thread_start.
To make my tests works I had to use https://github.com/meganz/mingw-std-threads.git and to modify c++config.h in mingw by commenting the line #define _GLIBCXX_HAS_GTHREADS 1.
Eventually, I had to do some magic in my CMakeList files to find mingw-std-threads implementation...
It's far from being straightforward.
It may be related to the the mrpt compile error (which is thread-related) but it's not clear why.
On your side, how do you make std::thread work with mingw 64?
huumm, thinking about it while writing...
SetThreadDescription is a win32 API?
my mingw 64 is posix.
I installed also a win32 but don't use it  because it does not implement std::thread (if I understand correctly the comparisons I found on the net).
So
Is it possible to build MRPT with mingw-64-posix, otherwise, do you know how to use mingw-64-win32 AND having std::thread?

Regards

closed 2024-12-25 by jlblancoc