MRPT/mrpt #1151: ndefined reference to symbol 'dlclose@@GLIBC_2.2.5' //lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line

BIT-glimmer, 2021-04-14:
[ 63%] Linking CXX executable ../../bin/DifOdometry-Datasets
/usr/bin/ld: //usr/local/lib/libglfw3.a(glx_context.c.o): undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
apps/DifOdometry-Datasets/CMakeFiles/DifOdometry-Datasets.dir/build.make:117: recipe for target 'bin/DifOdometry-Datasets' failed
make[2]: *** [bin/DifOdometry-Datasets] Error 1
CMakeFiles/Makefile2:4134: recipe for target 'apps/DifOdometry-Datasets/CMakeFiles/DifOdometry-Datasets.dir/all' failed
make[1]: *** [apps/DifOdometry-Datasets/CMakeFiles/DifOdometry-Datasets.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

jlblancoc, 2021-04-15:
The usual suspect here is anything under `/usr/local` like your `//usr/local/lib/libglfw3.a`... why do you have that library there? 
If possible, clean up everything under `/usr/local/`, delete mrpt's build directory and configure and build again, using system libraries only (those from `sudo apt install ...`).

BIT-glimmer, 2021-04-16:
Thanks a lot. I have built mrpt successfully with the system library but not on Ubuntu 16.04.

closed 2021-04-16 by BIT-glimmer