MRPT/mrpt #1159: How to install mrpt in windows10 and vs2019?

Tokymin, 2021-07-28:
I just install the windows .exe package,but it's said the path of  system System environment variable is too long. I also tried to compile the source code of the MRPT using cmake, but there are so error to solve, it just drive me crazy...T T

jlblancoc, 2021-07-29:
> ,but it's said the path of system System environment variable is too long

Hmm... Try the installer again **without** marking the option to "Add to system PATH".

Tokymin, 2021-07-29:
> 
> 
> > ,but it's said the path of system System environment variable is too long
> 
> Hmm... Try the installer again **without** marking the option to "Add to system PATH".

thanks for your reply,I've configured it correctly in the VS2019 this afternoon,^ ^

closed 2021-07-29 by jlblancoc