MRPT/mrpt #1161: Difodo datasets app doesn't work when using TUM datasets

e-sempertegui, 2021-09-07:
Hello, I've successfully built the MRPT library from source in Ubuntu18.04 (kernel 5.4.0-81) and now I'm trying to test the Difodo dataset app using the TUM datasets available [here](https://www.mrpt.org/Collection_of_Kinect_RGBD_datasets_with_ground_truth_CVPR_TUM_2011)  via the following command:

`DifOdometry-Datasets --config myconfig.txt`

However I keep getting the following exception and the program immediately crashes:
```
Loading dataset from: /home/emilk/Downloads/rawlog_rgbd_dataset_freiburg2_large_no_loop/rgbd_dataset_freiburg2_large_no_loop.rawlog
==== MRPT exception ====
Message:  Exception while parsing typed object 'CObservation3DRangeScan' from stream!
Original exception:
==== MRPT exception ====
Message:  Assert condition failed: rows > 0 && cols > 0
Location: /home/emilk/git/mrpt/libs/obs/src/CObservation3DRangeScan.cpp:444: [virtual void mrpt::obs::CObservation3DRangeScan::serializeFrom(mrpt::serialization::CArchive&, uint8_t)
Call stack backtrace:
[0 ] (unknown file) mrpt::obs::CObservation3DRangeScan::serializeFrom(mrpt::serialization::CArchive&, unsigned char)
[1 ] (unknown file) mrpt::serialization::CArchive::internal_ReadObject(mrpt::serialization::CSerializable*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, signed char)
[2 ] (unknown file) mrpt::serialization::CArchive::operator>>(std::shared_ptr<mrpt::serialization::CSerializable>&)
[3 ] (unknown file) mrpt::obs::CRawlog::loadFromRawLogFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
[4 ] (unknown file) CDifodoDatasets::loadConfiguration(mrpt::config::CConfigFileBase const&)
[5 ] (unknown file) main
[6 ] /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:344 __libc_start_main
[7 ] (unknown file) _start

Location: /home/emilk/git/mrpt/libs/serialization/src/CArchive.cpp:476: [void mrpt::serialization::CArchive::internal_ReadObject(mrpt::serialization::CSerializable*, const string&, bool, int8_t)
Call stack backtrace:
[0 ] (unknown file) mrpt::serialization::CArchive::internal_ReadObject(mrpt::serialization::CSerializable*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, signed char)
[1 ] (unknown file) mrpt::serialization::CArchive::operator>>(std::shared_ptr<mrpt::serialization::CSerializable>&)
[2 ] (unknown file) mrpt::obs::CRawlog::loadFromRawLogFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
[3 ] (unknown file) CDifodoDatasets::loadConfiguration(mrpt::config::CConfigFileBase const&)
[4 ] (unknown file) main
[5 ] /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:344 __libc_start_main
[6 ] (unknown file) _start

MRPT exception caught: ==== MRPT exception ====
Message:  Index out of bounds
Location: /home/emilk/git/mrpt/libs/obs/src/CRawlog.cpp:88: [mrpt::obs::CObservation::Ptr mrpt::obs::CRawlog::getAsObservation(size_t) const
Call stack backtrace:
[0 ] (unknown file) mrpt::obs::CRawlog::getAsObservation(unsigned long) const
[1 ] (unknown file) CDifodoDatasets::loadFrame()
[2 ] (unknown file) CDifodoDatasets::reset()
[3 ] (unknown file) main
[4 ] /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:344 __libc_start_main
[5 ] (unknown file) _start
```

I tried different sequences from the dataset, also tried using the PPAs versions from master and develop branches, yet I keep getting the same error. In addition, if I try analyzing the rawlogs using the command line tool 
`rawlog-edit --info -i ~/Downloads/rawlog_rgbd_dataset_freiburg2_large_no_loop/rgbd_dataset_freiburg2_large_no_loop.rawlog`
and I get almost the same error message
```
[rawlog-edit] Operation to perform: info
[rawlog-edit] Opening '/home/emilk/Downloads/rawlog_rgbd_dataset_freiburg2_large_no_loop/rgbd_dataset_freiburg2_large_no_loop.rawlog'...
[rawlog-edit] Open OK.
[rawlog-edit] Found external storage directory: /home/emilk/Downloads/rawlog_rgbd_dataset_freiburg2_large_no_loop/rgbd_dataset_freiburg2_large_no_loop_Images
[CRawlog::readActionObservationPair] Found exception:
==== MRPT exception ====
Message:  Exception while parsing typed object 'CObservation3DRangeScan' from stream!
Original exception:
==== MRPT exception ====
Message:  Assert condition failed: rows > 0 && cols > 0
Location: /home/emilk/git/mrpt/libs/obs/src/CObservation3DRangeScan.cpp:444: [virtual void mrpt::obs::CObservation3DRangeScan::serializeFrom(mrpt::serialization::CArchive&, uint8_t)
Call stack backtrace:
[0 ] (unknown file) mrpt::obs::CObservation3DRangeScan::serializeFrom(mrpt::serialization::CArchive&, unsigned char)
[1 ] (unknown file) mrpt::serialization::CArchive::internal_ReadObject(mrpt::serialization::CSerializable*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, signed char)
[2 ] (unknown file) mrpt::serialization::CArchive::operator>>(std::shared_ptr<mrpt::serialization::CSerializable>&)
[3 ] (unknown file) mrpt::obs::CRawlog::getActionObservationPairOrObservation(mrpt::serialization::CArchive&, std::shared_ptr<mrpt::obs::CActionCollection>&, std::shared_ptr<mrpt::obs::CSensoryFrame>&, std::shared_ptr<mrpt::obs::CObservation>&, unsigned long&)
[4 ] (unknown file) op_info(mrpt::io::CFileGZInputStream&, TCLAP::CmdLine&, bool)
[5 ] (unknown file) main
[6 ] /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:344 __libc_start_main
[7 ] (unknown file) _start

Location: /home/emilk/git/mrpt/libs/serialization/src/CArchive.cpp:476: [void mrpt::serialization::CArchive::internal_ReadObject(mrpt::serialization::CSerializable*, const string&, bool, int8_t)
Call stack backtrace:
[0 ] (unknown file) mrpt::serialization::CArchive::internal_ReadObject(mrpt::serialization::CSerializable*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, signed char)
[1 ] (unknown file) mrpt::serialization::CArchive::operator>>(std::shared_ptr<mrpt::serialization::CSerializable>&)
[2 ] (unknown file) mrpt::obs::CRawlog::getActionObservationPairOrObservation(mrpt::serialization::CArchive&, std::shared_ptr<mrpt::obs::CActionCollection>&, std::shared_ptr<mrpt::obs::CSensoryFrame>&, std::shared_ptr<mrpt::obs::CObservation>&, unsigned long&)
[3 ] (unknown file) op_info(mrpt::io::CFileGZInputStream&, TCLAP::CmdLine&, bool)
[4 ] (unknown file) main
[5 ] /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:344 __libc_start_main
[6 ] (unknown file) _start

Time to parse file (sec)          : 0.115348
Physical file size                : 127.78 KB
Uncompressed file size            : 115.00  B
Compression ratio                 : 111112.17%
Overall number of objects         : 1
Actions/SensoryFrame format       : No
Observations format               : No
All sensor labels                 :
```

Has anyone come across this before? Any idea what could be the problem and how to solve it?
Thanks a lot!

jlblancoc, 2021-09-13:
It was a bug, fixed in #1162. 
Please, try it after rebuilding the current `develop` branch.

e-sempertegui, 2021-09-14:
It works as expected now. Thank you for the support!

closed 2021-09-14 by e-sempertegui

jlblancoc, 2021-09-14:
Awesome!