MRPT/mrpt #1177: rawlog files not readable

christian-rauch, 2021-09-23:
Some rawlog files are not readable anymore.

1. download `rawlog_rgbd_dataset_freiburg1_360.tgz` from https://www.mrpt.org/Collection_of_Kinect_RGBD_datasets_with_ground_truth_CVPR_TUM_2011
2. run the `benchmarkingImageFeatures_GUI` example with that log file

This will print the errors:
```
QImage::scaled: Image is a null image
QImage::scaled: Image is a null image
QImage::scaled: Image is a null image
Exception message: Assert condition failed: rows > 0 && cols > 0
==== MRPT exception backtrace ====
[1] ../libs/obs/src/CObservation3DRangeScan.cpp:447: [virtual void mrpt::obs::CObservation3DRangeScan::serializeFrom(mrpt::serialization::CArchive&, uint8_t)] Assert condition failed: rows > 0 && cols > 0
[0] ../libs/serialization/src/CArchive.cpp:462: [void mrpt::serialization::CArchive::internal_ReadObject(mrpt::serialization::CSerializable*, const string&, bool, int8_t)] Exception while parsing typed object 'CObservation3DRangeScan' from stream!
```
and crash once trying to read the next image.

The first image might be indeed empty, but this used to work with older MRPT versions. It would be very useful to retain backward compatibility with log files.

jlblancoc, 2021-09-23:
Hi, 

Can you confirm what version of MRPT led to that error? In theory that exact regression was fixed in the `develop` branch version recently, confirmed reading with RawLogViewer, but I didn't try with `benchmarkingImageFeatures_GUI` so it might be a different issue (?)

christian-rauch, 2021-09-24:
I tested this on `2.1.3`. WIth the `devel` branch (822f8141a8d94d4946fac5a02d54b62a099adacf) I don't get all the errors, only:
```
QImage::scaled: Image is a null image
QImage::scaled: Image is a null image
QImage::scaled: Image is a null image
```
but the program still crashes when trying to access an image via "Next" button:
```c
#0  0x00007f33bdf4a574 in __GI___readdir64 (dirp=0x0) at ../sysdeps/posix/readdir.c:44
#1  0x000056487563fd84 in MainWindow::readFilesFromFolder(int) ()
#2  0x000056487564ad32 in MainWindow::on_next_button_clicked() ()
#3  0x00007f33bee04300 in QMetaObject::activate(QObject*, int, int, void**) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007f33bf8e5806 in QAbstractButton::clicked(bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#5  0x00007f33bf8e5a2e in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007f33bf8e6e73 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007f33bf8e7035 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007f33bf8332b6 in QWidget::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007f33bf7f0a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007f33bf7fa343 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007f33bedd893a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007f33bf7f9457 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007f33bf84f35d in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007f33bf8521ec in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007f33bf7f0a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007f33bf7fa0f0 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007f33bedd893a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007f33bf1c17d3 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#19 0x00007f33bf1c310b in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#20 0x00007f33bf19d35b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#21 0x00007f33aadfb32e in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#22 0x00007f33bb4d817d in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007f33bb4d8400 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#24 0x00007f33bb4d84a3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#25 0x00007f33bee30565 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007f33bedd74db in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007f33beddf246 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x0000564875635297 in main ()
```

I cannot build all of MRPT on the devel branch. I get the compilation error:
```
../apps/robot-map-gui/gui/CMainWindow.cpp: In member function ‘void CMainWindow::itemClicked(const QModelIndex&)’:
../apps/robot-map-gui/gui/CMainWindow.cpp:254:15: error: ‘CQtGlCanvasBase’ is not a member of ‘mrpt::gui’; did you mean ‘CGlCanvasBase’?
  254 |    mrpt::gui::CQtGlCanvasBase* gl = new mrpt::gui::CQtGlCanvasBase();
      |               ^~~~~~~~~~~~~~~
      |               CGlCanvasBase
```
so I am not certain that the crash might not come from this.

jlblancoc, 2021-09-27:
That build error on the `develop` branch is not found by neither CI nor my local builds... is Qt detected by cmake at configuration time??

On the OP: I can confirm the segfault, and started to partly fix it in [this branch](https://github.com/MRPT/mrpt/tree/feature/improve-benchmark-gui), but that app would need a serious refactoring work to get in shape, so I have to leave it for now... At least it now shows an exception error instead of crashing. 
The code must be refactored so images are actually grabbed from the .rawlog instead of reading them from a directory... the original implementation was kind of a hack, but didn't realize at the time (sigh). 

If anyone wants to give a hand, it would be more than welcome! ;-)

christian-rauch, 2021-09-27:
Alright. For what I originally intended to do, I now just use the `devel` branch to get the latest fix.

@jlblancoc Could you do a new release with that fix integrated, and also a new binary release on ROS/bloom with that fixed version?

christian-rauch, 2021-09-28:
> That build error on the `develop` branch is not found by neither CI nor my local builds... is Qt detected by cmake at configuration time??

Not sure what changed, but this build error went away.

jlblancoc, 2021-09-29:
> @jlblancoc Could you do a new release with that fix integrated, and also a new binary release on ROS/bloom (https://github.com/mrpt-ros-pkg-release/mrpt2-release) with that fixed version?

I will after release of v2.4.0 (soon, but a couple more weeks of use and testing is recommended to test the last minute features).

In the meanwhile, you could use the nightly binaries from [this PPA](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt).

closed 2024-07-27 by jlblancoc