MRPT/mrpt #1206: Remove more 3rd party sources from release tarballs

jlblancoc, 2021-12-13:
- [ ] 3rdparty/googletest/* -> libgtest-dev, libgmock-dev
- [ ] 3rdparty/nanogui/ext/nanovg/example/Roboto-*.ttf -> fonts-roboto
- [ ] 3rdparty/glew/* -> libglew-dev
- [ ] 3rdparty/freeglut/* -> freeglut3-dev
- [ ] 3rdparty/glext/* -> libgl-dev
- [ ] libs/math/src/CSparse/ -> libsuitesparse-dev

As suggested by @jspricke .

Alternatives (on the fly download / clone?) should be added for Windows users, I guess...

jspricke, 2021-12-14:
>Alternatives (on the fly download / clone?) should be added for Windows users, I guess...

There are package managers for Windows, maybe they could provide the 
dependencies?

jlblancoc, 2021-12-14:
@jspricke well, thanks to the d/copyright Files-Excluded feature, it was far easier than expected... only one patch needed for gtest, the rest are handled via d/copyright.

closed 2021-12-14 by jlblancoc