MRPT/mrpt #1225: Question about the distorsion coefficient in the yml file

PotatoDroid5, 2022-05-03:
Hello, I need to convert the intrinsic parameters from .yml to .cam format. Just to be sure, in the yml file generated:

camera_matrix:
  cols: 3
  data: [652.3032180347936, 0, 324.1105454718237, 0, 647.8789451143271, 235.0098017548972, 0, 0, 1]
  rows: 3
camera_name: camera1
distortion_coefficients:
  cols: 8
  data: [-0.1497031993890412, 1.362350091825391, 0.003514720345557172, 0.008293638896066356, -3.412437392789501, 0, 0, 0]
  rows: 1

is it equivalent to:
  <focallength x="652.3032180347936" y="647.8789451143271" />
  <principalpoint x="324.1105454718237" y="235.0098017548972" />
  <distortion k1="-0.1497031993890412" k2="1.362350091825391" k3="0.003514720345557172" p1="0.008293638896066356" p2="-3.412437392789501" />

or is the order of the distorsion coefficients different?

jlblancoc, 2022-05-03:
Hi @PotatoDroid5 : 

I don't know the .cam format, please give some pointers to their docs... 
In the MRPT-produced .yml file, the order of distortion parameters is compatible with that of opencv and/or ROS. 
Check the "D" entry here: http://docs.ros.org/en/api/sensor_msgs/html/msg/CameraInfo.html
or the distortions in this answer: https://stackoverflow.com/a/34024057

PotatoDroid5, 2022-05-04:
Hi @jlblancoc 

The thing is, I don't know. It was a file format already used in a project I had to work on when I joined my current company, and my predecessor left several years ago without leaving much documentation behind. When looking more into details, it may be a format he made himself. It relies on xml entities with pretty straightforward names, like principalpoint x="..." y="..." or distortion k1="..."... The xml entities didn't show up in my 1st post, so I can understand that the question may have seem too vague (it should appear correctly now).

To reword the question: in the yml format, is the order of the distortion coefficient [k1, k2, k3, p1, p2,...]? In the link you sent, it seems to be [k1, k2, p1, p2, k3, k4, k5, k6], is that right?

Thanks for your answer and the links

PotatoDroid5, 2022-05-05:
Thank for the confirmation.

Have a nice day :)

closed 2022-05-05 by PotatoDroid5