MRPT/mrpt #1232: pymrpt.so in in python sys.path

rdinoff, 2022-06-01:
after installing ros-noetic-mrpt-bridge, pymrpt.so is placed in the wrong directory. The file needs to be in the pythonpath:

```
$ dpkg -S pymrpt
ros-noetic-mrpt2: /opt/ros/noetic/lib/x86_64-linux-gnu/python3.8/dist-packages/pymrpt.so

$ dpkg -l ros-noetic-mrpt2
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name             Version                      Architecture Description
+++-================-============================-============-===================================================
ii  ros-noetic-mrpt2 2.4.4-1focal.20220514.015241 amd64        Mobile Robot Programming Toolkit (MRPT) version 2.x

$ python
Python 3.8.10 (default, Mar 15 2022, 12:22:08) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pymrpt
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pymrpt'
>>> import sys
>>> print(sys.path)
['', '/opt/ros/noetic/lib/python3/dist-packages', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']
```

jlblancoc, 2022-06-02:
Good point. Will try to fix it for the next release. 

Any further testing or improvements in the python wrappers will be welcome, since we normally only focus on C++ ;-)

jlblancoc, 2022-06-06:
Closed via 5486914e5635da8839f2a558f72ccf533ef76750

closed 2022-06-06 by jlblancoc