MRPT/mrpt #1233: pymrpt.so already registered; second conversion method ignored.

rdinoff, 2022-06-01:
on noetic if I link the pymrpt.so so that it is in the pythonpath many warnings are generated.  The library does work.  

```
$ ln -s /opt/ros/noetic/lib/x86_64-linux-gnu/python3.8/dist-packages/pymrpt.so /usr/local/lib/python3.8/dist-packages/pymrpt.so

$ python
Python 3.8.10 (default, Mar 15 2022, 12:22:08) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pymrpt
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::poses::CPose2D> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::poses::CPosePDF> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::poses::CPosePDFGaussian> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::poses::CPosePDFParticles> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::poses::CPose3D> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::poses::CPose3DPDF> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::poses::CPose3DPDFGaussian> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::poses::CPose3DPDFParticles> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::obs::CAction> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::obs::CActionRobotMovement2D> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::obs::CObservation> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::obs::CObservationOdometry> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::obs::CObservationRange> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::obs::CObservation2DRangeScan> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::obs::CObservationBearingRange> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::obs::CSensoryFrame> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::maps::TMapGenericParams> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::maps::CMetricMap> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::maps::COccupancyGridMap2D> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::maps::CPointsMap> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::maps::CSimplePointsMap> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::maps::CSimpleMap> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::maps::CMultiMetricMap> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::maps::CMultiMetricMapPDF> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::opengl::CRenderizable> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::opengl::CGridPlaneXY> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::opengl::CSetOfObjects> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::opengl::CSetOfLines> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::opengl::CEllipsoid3D> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::opengl::CEllipsoid2D> already registered; second conversion method ignored.
<frozen importlib._bootstrap>:219: RuntimeWarning: to-Python converter for std::shared_ptr<mrpt::opengl::COpenGLScene> already registered; second conversion method ignored.

```

jlblancoc, 2022-06-02:
Note: Potential workaround [here](https://stackoverflow.com/a/13017303)

jlblancoc, 2023-06-20:
This problem affected mrpt<2.9.0, so it's over now.

closed 2023-06-20 by jlblancoc