MRPT/mrpt #1234: 2.4.8 source fails to build on Fedora (missing files)

git-user44, 2022-06-02:
Trying to build MRPT from source on Fedora (the version in our repos is 1.4).

mrpt-2.4.8/libs/config/src/CConfigFile.cpp references SimpleIni.h.

I have had a good look about in the Fedora repos, but there is nothing providing this file.

Looking around here I see a source file for MRPT-1.5.5. Not sure if this is right or not.

jlblancoc, 2022-06-03:
Hi, 

`SimpleIni` is a required build dependency, which can be provided from either its own "copy" in MRPT source tree via a git submodule (under [3rdparty/simpleini](https://github.com/MRPT/mrpt/tree/develop/3rdparty) ), or via a system package.

Since `libsimpleini-dev` exists in Debian & modern Ubuntu repos, that submodule is *not* shipped in source tarballs (that's handled via [this line in this script](https://github.com/MRPT/mrpt/blob/develop/packaging/make_release.sh#L92)).
As you can see there, one could run (on a git clone of this repo): 

```bash
git checkout 2.4.8
MRPT_PKG_EXPORTED_SUBMODULES=simpleini ./packaging/make_release.sh
```

and you will obtain an alternative tarball *with* simpleini, which could be packaged for Fedora, if there are no plans to integrate this library in Fedora in the short term...

Any other ideas or alternatives that make packaging easier for the future are welcome! ;-)

git-user44, 2022-06-04:
Thanks for the reply. I guess that explains why nanoflann didn't exist either.

I'm trying to make MRPT a package for use within buildroot, to make cross compiling/deployment (possibly) easier. Host machine is X86_64 running Fedora 35 and the target robot is aarch64 running a buildroot system. Slowly getting there.

So then, back 2 or 3 steps and see if I can do this properly using MRPT_PKG_EXPORTED_SUBMODULES.

closed 2022-06-04 by git-user44

jlblancoc, 2022-06-06:
Oh, I see. If you finally managed to do it and want to contribute a note on how to do it, feel free of opening a PR to document it, or report it here and I'll format it. 
Best.