MRPT/mrpt #1235: Missing file or naming problem - I am not sure

git-user44, 2022-06-04:
cmake_install.cmake references a file called "mrpt-2.4.8/parse-files/ep_eigen3-config.cmake"

it does not exist.

I assume it should really be "mrpt-2.4.8/parse-files/EP_eigen3-config.cmake"

or maybe the cmake file is right and the file is wrongly named. I don't know.

jlblancoc, 2022-06-06:
Thanks for reporting. 

It was a complicated issue related to not having a system's provided version of Eigen3.

It seems to be fixed now after the commit mentioned above (pushed to develop branch), but please give it a try and reopen if it persists.

If it works, we'll make a new release soon, suitable for packaging.

git-user44, 2022-06-07:
Thanks.

I can confirm those changes work.

jlblancoc, 2022-06-07:
Awesome!

commit 239ebaf6 (2022-06-06): Fix handling of embedded Eigen (Closes #1235)