MRPT/mrpt #1251: When configuring the running environment of pl-slam, install mrpt software, and then this error occurs

xqs-c, 2022-09-03:
When configuring the running environment of pl-slam, install mrpt software, and then this error occurs。Mrpt version is https://github.com/MRPT/mrpt/tree/0c3d605c3cbf5f2ffb8137089e43ebdae5a55de3

error:
/usr/bin/ld: /usr/local/wxWidgets/lib/libwx_baseu-3.1.a(baselib_threadinfo.o): relocation R_X86_64_TPOFF32 against `_ZZN12_GLOBAL__N_117GetThisThreadInfoEvE16s_thisThreadInfo' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/local/wxWidgets/lib/libwx_baseu-3.1.a(baselib_utilsunx.o): relocation R_X86_64_PC32 against symbol `_ZN16wxAppConsoleBase14ms_appInstanceE' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: The last link failed: wrong value
collect2: error: ld returned 1 exit status
libs/base/CMakeFiles/mrpt-base.dir/build.make:3638: recipe for target 'lib/libmrpt-base.so.1.5.0' failed
make[2]: *** [lib/libmrpt-base.so.1.5.0] Error 1
CMakeFiles/Makefile2:259: recipe for target 'libs/base/CMakeFiles/mrpt-base.dir/all' failed
make[1]: *** [libs/base/CMakeFiles/mrpt-base.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

ubuntu 18.04
wxWidgets3.1.0
opencv3.1.0
gcc7.5.0

jlblancoc, 2022-09-09:
Please, give a try to the code in this PR, which was never merged:
https://github.com/rubengooj/pl-slam/pull/40

jolting, 2022-09-09:
```
/usr/local/wxWidgets/lib/libwx_baseu-3.1.a
```
Local build of wx built without fPIC. 

PIC wasn't enabled by default until 3.1.5
https://github.com/wxWidgets/wxWidgets/commit/4ee04daa7ceb758ba8c67fb9e01528605e4fc2da

You have to build it with `--with-pic`.

closed 2024-07-27 by jlblancoc