MRPT/mrpt #1258: [Help, please] Problems building source code on Windows

phamhoanganhbk, 2022-10-28:
Hi everyone,
I tried to build source code on Windows. (specifically the code sample `slam_icp_simple_example`). 
Step 1: Download source code 
`git clone https://github.com/MRPT/mrpt.git --recursive` 
Step 2: Use CMake GUI to generate project files for VS2022
But it always has GLFW library path problem even though I tried many ways to install it.
`Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.22000.
CMake Error at C:/Program Files/CMake/share/cmake-3.25/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "Findglfw.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "glfw", but
  CMake did not find one.
  Could not find a package configuration file provided by "glfw" with any of
  the following names:
    glfwConfig.cmake
    glfw-config.cmake
  Add the installation prefix of "glfw" to CMAKE_PREFIX_PATH or set
  "glfw_DIR" to a directory containing one of the above files.  If "glfw"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  C:/Program Files/mrpt-2.5.6/lib/cmake/mrpt-nanogui/mrpt-nanogui-config.cmake:43 (find_dependency)
  C:/Program Files/CMake/share/cmake-3.25/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  C:/Program Files/mrpt-2.5.6/lib/cmake/mrpt-gui/mrpt-gui-config.cmake:43 (find_dependency)
  CMakeLists.txt:26 (find_package)

Configuring incomplete, errors occurred!
See also "D:/github/mrpt/samples/slam_icp_simple_example/build/CMakeFiles/CMakeOutput.log".`

Could you give me some advice, please ? (I have successfully tested it on Ubuntu)
Many thanks,

jlblancoc, 2022-10-28:
Hi, 

If you git cloned the sources, why cmake detecting what seems to be an already built version under `C:/Program Files/mrpt-2.5.6/...` ?? 

If you formerly installed mrpt in Windows with the installer, please uninstall it first (through the usual "Control Panel -> Apps" ), remove the cmake build directory, open cmake-gui again, select as source dir the place where you cloned the repo, and as build directory a new, empty directory somewhere else, and try again... 

It's supposed to build ok, including nanogui, under Windows on our [CI pipeline](https://ci.appveyor.com/project/jlblancoc/mrpt/builds/45168621?fullLog=true) so it should work... let me know if you still have problems after following those steps.

phamhoanganhbk, 2022-10-28:
Hi @jlblancoc 
Thank you for your reply.
I uninstalled mrpt installer and then i removed the cmake build directory of `slam_icp_simple_example` (because i just want to run the sample, i don't know if i need to build the mrpt project first?).
Then it came up with another error

`Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.22000.
The C compiler identification is MSVC 19.33.31630.0
The CXX compiler identification is MSVC 19.33.31630.0
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.33.31629/bin/Hostx64/x64/cl.exe - skipped
Detecting C compile features
Detecting C compile features - done
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.33.31629/bin/Hostx64/x64/cl.exe - skipped
Detecting CXX compile features
Detecting CXX compile features - done
CMake Error at CMakeLists.txt:26 (find_package):
  By not providing "Findmrpt-slam.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "mrpt-slam", but CMake did not find one.

  Could not find a package configuration file provided by "mrpt-slam" with
  any of the following names:

    mrpt-slamConfig.cmake
    mrpt-slam-config.cmake

  Add the installation prefix of "mrpt-slam" to CMAKE_PREFIX_PATH or set
  "mrpt-slam_DIR" to a directory containing one of the above files.  If
  "mrpt-slam" provides a separate development package or SDK, be sure it has
  been installed.
Configuring incomplete, errors occurred!
See also "D:/github/mrpt/samples/slam_icp_simple_example/build/CMakeFiles/CMakeOutput.log".`

Many thanks,

phamhoanganhbk, 2022-11-01:
Hi,
I have solved the problem.
For those new (like me) on Windows who want to test the samples
Step 1: Download source code (not Installer)
Step 2: CMake GUI, follow the author's instructions. But pay attention to click on MRPT_BUILD_EXAMPLES (I forgot in this step).
Step 3: Link *.dll libraries to sample projects.

closed 2022-11-01 by phamhoanganhbk