MRPT/mrpt #1270: New variance field for the Range message

ejalaa12, 2023-03-13:
The `Range` message will get an additional field for the variance.
I don't know if this is relevant for you guys, but this issue is simply a notification, feel free to close it if it's not a change that you could benefit from.

You can see details about the PR [here](https://github.com/ros2/common_interfaces/pull/181)

jlblancoc, 2023-10-23:
@ejalaa12 This has been finally addressed here: 
 https://github.com/MRPT/mrpt/commit/67358e90876f6af1b1021ec03d60740c68521157

so MRPT >= v2.11.2 will use the new variance field.

ejalaa12, 2023-10-23:
@jlblancoc awesome! I saw the commit, this a clever way of making sure the member exist.
I guess this makes it retro-compatibke with old version of the Range message as well.

jlblancoc, 2023-10-24:
Exactly 👍

jlblancoc, 2023-10-24:
However, for some reason, the `if constexpr` trick fails on Humble... hmm....

https://build.ros2.org/job/Hdev__mrpt2__ubuntu_jammy_amd64/349/console

```
03:46:44 [ 92%] Building CXX object libs/ros2bridge/CMakeFiles/ros2bridge.dir/src/range.cpp.o
03:46:44 /tmp/ws/src/mrpt2/libs/ros2bridge/src/range.cpp: In function ‘bool mrpt::ros2bridge::fromROS(const Range&, mrpt::obs::CObservationRange&)’:
03:46:44 /tmp/ws/src/mrpt2/libs/ros2bridge/src/range.cpp:43:78: error: ‘const Range’ {aka ‘const struct sensor_msgs::msg::Range_<std::allocator<void> >’} has no member named ‘variance’
03:46:44    43 |                 obj.sensedData.at(0).sensorNoiseStdDeviation = std::sqrt(msg.variance);
03:46:44       |                                                                              ^~~~~~~~

``` 

I'll investigate it.

jolting, 2023-10-24:
This was added on March 15. It didn't make Humble.
https://github.com/ros2/common_interfaces/commit/05d7e1942acffefbeb40f7dd9d69ce4298119dcf

jolting, 2023-10-24:
Maybe try it with declval?
```
template <class T>
struct has_variance<T, std::void_t<decltype(std::declval<T>().variance)>> : std::true_type
{
};
```

jlblancoc, 2023-10-25:
> This was added on March 15. It didn't make Humble. [ros2/common_interfaces@05d7e19](https://github.com/ros2/common_interfaces/commit/05d7e1942acffefbeb40f7dd9d69ce4298119dcf)

Yeap, exactly! :-)

jlblancoc, 2023-10-25:
@jolting It was something else... I just found (the hard way..) that everything inside an `if constexpr()` **is** actually parsed by the compiler (even if the condition is false)... **except** if it's inside a template. Solution -> move that part to a template, and it's now working (ignoring) on humble too! https://github.com/MRPT/mrpt/commit/4b6e10895848a1c66e12d26e63dad4e3ebb52bc8

commit 67358e90 (2023-10-23): Address new field variance in ROS msg Range (Closes #1270)
commit 4b6e1089 (2023-10-25): Fix correct use of if constexpr()

Closes #1270
closed 2023-10-25 by jlblancoc