MRPT/mrpt #1275: pymrpt broken in ros-noetic-mrpt2 in 2.9.0

ihadzic, 2023-06-16:
`ros-noetic-mrpt2` package just got an update to 2.9.0 and `pymrpt` broke there. I can have an import like this:

```
from pymrpt.poses import CPose3DPDFGaussian
```

and it pukes with this message:

```
from pymrpt.poses import CPose3DPDFGaussian
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pymrpt'
```

This has worked fine in 2.8.0 and the bindings were found in `/opt/ros/noetic/lib/python3/dist-packages/pymrpt.so`. That file is gone in 2.9.0 but in instead what showed up that looks like should be Python binding are files under `/opt/ros/noetic/lib/python3.8/site-packages/mrpt` but that doesn't import and even if I try to hack my way into importing it doesn't look like a correct `pymrpt` module (it doesn't have `poses` submodule at all).

jlblancoc, 2023-06-17:
Ok, I see. I have verified it on a u20.04 system with the latest packages from the ROS build farm for Noetic.

Two different things here: 

1) python3 not finding the library. That's "easy" to hack away, as you probably have already done, either creating symlinks under `/opt/ros/noetic/lib/python3/dist-packages` (not recommended) or by temporarily adding: 

        export PYTHONPATH=$PYTHONPATH:/opt/ros/noetic/lib/python3.8/site-packages/

2) With MRPT 2.9.0, there has been a massive change in the Python bindings: it formerly was hand-written and only exported part of the API. It's now automated via pybind11 and most parts of MRPT are now available via `pymrpt`. The names and such changed a bit, though. 
Please, see:
    - Examples like [se3-poses-example.py](https://github.com/MRPT/mrpt/blob/develop/python-examples/se3-poses-example.py) or [hwdriver-tao-imu-usb.py](https://github.com/MRPT/mrpt/blob/develop/python-examples/hwdriver-tao-imu-usb.py) for a possible way to import the module (first few lines), and
    - The only (ugly) [pydoc reference docs](https://mrpt.github.io/pymrpt-docs/mrpt.pymrpt.mrpt.html) of all existing classes, modules and methods. If you use the interactive python3 interpreter, `tab` will help you to autocomplete, but I wasn't able to make Visual Studio Code to autocomplete pymrpt classes, and couldn't find the reason...

Any help with testing, improving, adding/correcting Python examples, etc. would be more than welcome, since we normally develop with the C++ libs and the python part is far less tested...

I'm leaving this ticket open to remember trying to fix the installation path of pymrpt in noetic.

jlblancoc, 2023-06-17:
PS: I realized the logic to determine the install directory for pymrpt is inverted! `python3` is used for ROS2, `python3.X` for ROS1, while it should be the other way around. 
I'll fix it asap.

ihadzic, 2023-06-20:
Thanks for fixing this. I am still struggling to find everything I need in the new API. Specifically, I need to go back and forth between ROS1 messages and MRPT types. In the old API I would use something like this:

```
p1mrpt = CPose3DPDFGaussian()
p1mrpt.from_ROS_PoseWithCovariance_msg(p1)
```

where `p1` is `PoseWithCovariance` type from `geometry_msgs.msg`

In the new API, I understand I should use `ros1bridge` that is documented here:

https://docs.mrpt.org/reference/latest/group_mrpt_ros1bridge_grp.html

and call `toROS` and `fromROS` to do the conversion. Assuming that Python bindings follow the hierarchy that C++ follows, I would expect to find `ros1bridge` under `pymrpt.mrpt`, but I don't see it:

```
>>> from mrpt import pymrpt
>>> dir(pymrpt.mrpt)
['Clock', 'DEG2RAD', 'RAD2DEG', 'Stringifyable', 'TCallStackBackTrace', 'TCallStackEntry', 'WorkerThreadsPool', '__doc__', '__loader__', '__name__', '__package__', '__spec__', 'aligned_calloc', 'aligned_free', 'aligned_malloc', 'apps', 'bayes', 'callStackBackTrace', 'comms', 'config', 'containers', 'cpu', 'd2f', 'expr', 'f2u8', 'global_settings', 'graphs', 'gui', 'hwdrivers', 'hypot_fast', 'ignored_copy_ptr_mrpt_maps_COctoMapBase_octomap_ColorOcTree_octomap_ColorOcTreeNode_t', 'ignored_copy_ptr_mrpt_maps_COctoMapBase_octomap_OcTree_octomap_OcTreeNode_t', 'img', 'int_select_by_bytecount_1_t', 'int_select_by_bytecount_2_t', 'int_select_by_bytecount_3_t', 'int_select_by_bytecount_4_t', 'int_select_by_bytecount_8_t', 'io', 'keep_max', 'keep_min', 'kinematics', 'maps', 'math', 'nav', 'non_copiable_ptr_basic_void_t', 'obs', 'opengl', 'poses', 'ptr_cast_mrpt_serialization_CSerializable_t', 'random', 'round', 'rtti', 'safe_ptr_basic_mrpt_opengl_Scene_t', 'safe_ptr_basic_mrpt_rtti_TRuntimeClassId_t', 'safe_ptr_mrpt_opengl_Scene_t', 'safe_ptr_mrpt_rtti_TRuntimeClassId_t', 'serialization', 'sign', 'signWithZero', 'slam', 'square', 'system', 'tfest', 'toNativeEndianness', 'to_string', 'topography', 'typemeta', 'u8tof', 'uint_select_by_bytecount_1_t', 'uint_select_by_bytecount_2_t', 'uint_select_by_bytecount_3_t', 'uint_select_by_bytecount_4_t', 'uint_select_by_bytecount_8_t', 'vision']
```

I do see other modules that I recognize from docs (e.g. `poses`, `slam`, etc..), but no `ros1bridge`. Am I missing something or has this binding been omitted?

jlblancoc, 2023-06-20:
Oh my! You are right, it was totally left out (!). 

Following up in #1276.