MRPT/mrpt #1282: ros-noetic-mrpt2 dependencies

ihadzic, 2023-08-11:
The `ros-noetic-mrpt2` package seems to be rather heavy on dependencies. It pulls various QT and other GUI-related packages, which then start pulling all kinds of desktop packages.

On many installations these are unnecessary. If running on the robot (i.e. an embedded system) which has no display and runs minimal installation of Ubuntu, these extra dependencies are just clutter. I am assuming that these GUI packages are necessary if one is using visualization features inherent to MRPT, but ROS has its own visualization and debugging tools which typically run outside the robot and communicate the data.

How hard would it be to separate the packages in say ros-noetic-mrpt2 and ros-noetic-mrpt-2-core, where the latter would be just the guts of the math-crunching functions and the former would be for those who need full features of MRPT (presumably on desktop). The latter could be a dependency of the former?

jlblancoc, 2023-08-13:
That makes a lot of sense, and I'm aware of the lot of dependencies implied by installing mrpt2... It's not been done yet because it's far from trivial, here upstream, and from the ROS packaging itself.

Unlike Debian / Ubuntu, ROS build farms are prepared to create one single .deb file for each ROS package, that's why everything is together right now. 

(Totally) removing all the GUI and opengl stuff and then adding a second package that includes the optional GUI stuff is not doable since mrpt-gui and mrpt-opengl are "core" parts...

Something that has been in my mind for a long time, though, is creating a separate ROS package for the libraries, and another for the applications only.

Dropping the whole Qt might be accepted if we leave out the unique app that depends on it ("robot-map-gui").

I'll try to investigate an idea: 
- [ ] Remove the root `package.xml`
- [ ] Create a couple of "dummy" package subdirectories, each generating ROS packages with (1) the libraries, and (2) the apps.
- [ ] If it works, make a new release.

jlblancoc, 2024-07-27:
For the records: It was about time, but the good way to go was indeed splitting the ROS package `mrpt2` into more modular, smaller components. 

It's now done with this repo: https://github.com/MRPT/mrpt_ros
Once all active ROS 2 distributions go over one update cycle and the [new replacements](https://github.com/MRPT/mrpt_ros?tab=readme-ov-file#mrpt-ros) are available, `mrpt2` will be removed.

closed 2024-07-27 by jlblancoc