MRPT/mrpt #1317: Linking error in building apps.

bchoineubility, 2024-06-21:
Hi community,

I am brandnew for mrpt2 and trying to build mrpt2 for humble edition now.  by the way, I have an issue for building apps which include header file, libs/gui/include/mrpt/3rdparty/mathplot/mathplot.h.

error log tell `undefined reference to mpMovableObject::~mpMovableObject()`.

error log is the followinng:

```
| [34/221] : && /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe4t-linux/aarch64-oe4t-linux-g++ --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot -march=armv8-a+crc -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -DNDEBUG  -march=armv8-a+crc -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now    -L=/usr/local/cuda-11.4/lib -L=/usr/local/cuda-11.4/lib/stubs   -Wl,-rpath-link,/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot/usr/local/cuda-11.4/lib   -Wl,-rpath,/usr/local/cuda-11.4/lib  --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now    -L=/usr/local/cuda-11.4/lib -L=/usr/local/cuda-11.4/lib/stubs   -Wl,-rpath-link,/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot/usr/local/cuda-11.4/lib   -Wl,-rpath,/usr/local/cuda-11.4/lib apps/2d-slam-demo/CMakeFiles/2d-slam-demo.dir/slamdemoApp.cpp.o apps/2d-slam-demo/CMakeFiles/2d-slam-demo.dir/slamdemoMain.cpp.o apps/2d-slam-demo/CMakeFiles/2d-slam-demo.dir/batch_experiments.cpp.o apps/2d-slam-demo/CMakeFiles/2d-slam-demo.dir/CDlgParams.cpp.o apps/2d-slam-demo/CMakeFiles/2d-slam-demo.dir/CLogView.cpp.o -o bin/2d-slam-demo -L/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot/usr/local/cuda-11.4/lib64 -Wl,-rpath,"/usr/local/cuda-11.4/lib64:\$ORIGIN/../lib:"  lib/libmrpt-gui.so.2.10.2  lib/libmrpt-slam.so.2.10.2  -L/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot/usr/lib  -pthread  -lwx_baseu-3.1  -lwx_gtk3u_core-3.1  -lwx_gtk3u_gl-3.1  -lwx_gtk3u_aui-3.1  -lwx_gtk3u_html-3.1  lib/libmrpt-nanogui.so.2.10.2  -lglfw  lib/libmrpt-vision.so.2.10.2  lib/libmrpt-maps.so.2.10.2  lib/libmrpt-obs.so.2.10.2  lib/libmrpt-tfest.so.2.10.2  lib/libmrpt-graphs.so.2.10.2  lib/libmrpt-opengl.so.2.10.2  lib/libmrpt-poses.so.2.10.2  lib/libmrpt-bayes.so.2.10.2  lib/libmrpt-img.so.2.10.2  lib/libmrpt-io.so.2.10.2  lib/libmrpt-math.so.2.10.2  lib/libmrpt-serialization.so.2.10.2  lib/libmrpt-rtti.so.2.10.2  lib/libmrpt-random.so.2.10.2  lib/libmrpt-config.so.2.10.2  lib/libmrpt-expr.so.2.10.2  lib/libmrpt-system.so.2.10.2  lib/libmrpt-containers.so.2.10.2  lib/libmrpt-core.so.2.10.2 && :
| FAILED: bin/2d-slam-demo
| : && /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot-native/usr/bin/aarch64-oe4t-linux/aarch64-oe4t-linux-g++ --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot -march=armv8-a+crc -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -DNDEBUG  -march=armv8-a+crc -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0=/usr/src/debug/mrpt2/2.10.2-1-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10.2-1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a_tegra234-oe4t-linux/mrpt2/2.10. [... cut ...] /slamdemoMain.cpp.o: in function `mpPolygon::mpPolygon(wxString const&)':
| /usr/src/debug/mrpt2/2.10.2-1-r0/git/libs/gui/include/mrpt/3rdparty/mathplot/mathplot.h:1881: undefined reference to `mpMovableObject::~mpMovableObject()'
| collect2: error: ld returned 1 exit status
```

I have to do workarround by removing apps which include the header file in the `apps/CMakeLists.txt` as follows:
```
list(REMOVE_ITEM LIST_MRPT_APP_DIRS  2d-slam-demo RawLogViewer)
```

Can you please help me?

BR,
Mark

jlblancoc, 2024-06-21:
Are you using the [standard way](https://docs.mrpt.org/reference/latest/compiling.html#build-using-cmake) to build using cmake? 

What is your OS? 

Anyway, I just [pushed a commit](https://github.com/MRPT/mrpt/commit/d83b49b1395721faea5215c3da63337db2f02653) that might fix it, but it's really weird that this error never happened to us or to any of the CI builds :-) So, if you have cloned the repo, please do "git pull" for the develop branch and try again.

bchoineubility, 2024-06-23:
@jlblancoc Many thanks for the help 

I will test your push and get back to you.

Thanks again.  👍 

BR,
Mark

bchoineubility, 2024-06-24:
@jlblancoc , I can build all apps without issue. 

Thanks.

BR,
Mark

closed 2024-06-24 by bchoineubility

robwoolley, 2024-07-03:
Thanks @jlblancoc and @bchoineubility!  I ran into this when building with https://github.com/ros/meta-ros.  I can confirm that this fixes the issue.

The version of gcc was: gcc version 14.1.0 (GCC).  There were no special compiler options in play that I could tell, I managed to reproduce it simply with:

aarch64-oe-linux-g++   --sysroot=/path/to/sysroot -mcpu=cortex-a72+crc   apps/2d-slam-demo/CMakeFiles/2d-slam-demo.dir/slamdemoApp.cpp.o apps/2d-slam-demo/CMakeFiles/2d-slam-demo.dir/slamdemoMain.cpp.o apps/2d-slam-demo/CMakeFiles/2d-slam-demo.dir/batch_experiments.cpp.o apps/2d-slam-demo/CMakeFiles/2d-slam-demo.dir/CDlgParams.cpp.o apps/2d-slam-demo/CMakeFiles/2d-slam-demo.dir/CLogView.cpp.o -o bin/2d-slam-demo    -L/ala-lpggp31/rwoolley/pr1144/build/tmp-glibc/work/cortexa72-oe-linux/mrpt2/2.13.1-1/build/lib  -L/ala-lpggp31/rwoolley/pr1144/build/tmp-glibc/work/cortexa72-oe-linux/mrpt2/2.13.1-1/recipe-sysroot/usr/lib  -pthread  -lwx_baseu-3.2  -lwx_gtk3u_core-3.2  -lwx_gtk3u_gl-3.2  -lwx_gtk3u_aui-3.2  -lwx_gtk3u_html-3.2  lib/libmrpt-nanogui.so.2.13.1  -lglfw  -lmrpt-vision -lmrpt-obs -lmrpt-opengl -lmrpt-img -l mrpt-io -lmrpt-poses -lmrpt-math -lmrpt-serialization -lmrpt-random -lmrpt-config -lmrpt-system -lmrpt-core -lmrpt-gui -lmrpt-slam

jlblancoc, 2024-07-03:
Awesome! Thanks for the additional feedback, @robwoolley :+1: 
So, it seems our (and OSRF's ROS) CI has a blind spot regarding testing for Yocto... :-) 

For the records: Using MRPT >= v2.13.2 should avoid the error.

bchoineubility, 2024-07-03:
@robwoolley 👍