MRPT/mrpt #1321: can't find ros2bridge lib

guyupan1911, 2024-09-05:
CMakeLists.txt
` find_package(mrpt-ros2bridge)
target_link_libraries(${PROJECT_NAME}
  mrpt::ros2bridge
)
`
export error
`CMake Warning at modules/ros2_bridge/CMakeLists.txt:5 (find_package):
  By not providing "Findmrpt-ros2bridge.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "mrpt-ros2bridge", but CMake did not find one.

  Could not find a package configuration file provided by "mrpt-ros2bridge"
  with any of the following names:

    mrpt-ros2bridgeConfig.cmake
    mrpt-ros2bridge-config.cmake

  Add the installation prefix of "mrpt-ros2bridge" to CMAKE_PREFIX_PATH or
  set "mrpt-ros2bridge_DIR" to a directory containing one of the above files.
  If "mrpt-ros2bridge" provides a separate development package or SDK, be
  sure it has been installed.
`

jlblancoc, 2024-09-05:
Hi, 
Can you provide more context on this error? 

If it happens for a ROS2 package, and MRPT was installed via "apt", note that since ~1 month ago the `mrpt2` package is deprecated and you would need to declare dependencies following [this table](https://github.com/MRPT/mrpt_ros?tab=readme-ov-file#mapping-between-ros-packages--mrpt-c-libraries).

Example from the mola_bridge_ros2 package: 
- cmake lines: https://github.com/MOLAorg/mola/blob/develop/mola_bridge_ros2/CMakeLists.txt#L19-L21
- package.xml lines: https://github.com/MOLAorg/mola/blob/develop/mola_bridge_ros2/package.xml#L31-L32

guyupan1911, 2024-09-05:
Thanks for reply promptly, yes i am using mrpt-ros2bridge in a ros2 package to convert mrpt message to ros2, and the mrpt was installed via "apt".
one question is , i can find 
```
libmrpt-ros1bridge-dev:amd64
libmrpt-ros1bridge2.13:amd64   
```
but  there is no something like `libmrpt-ros2bridge`
also , i follow your example in mola-bridge_ros2 package to install mrpt_ros packages, but meet other problems when run `rosdep install..`

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  ros-humble-mrpt-libbase ros-humble-mrpt-libmaps ros-humble-mrpt-libmath ros-humble-mrpt-libobs ros-humble-mrpt-libopengl
  ros-humble-mrpt-libposes
The following NEW packages will be installed:
  ros-humble-mrpt-libbase ros-humble-mrpt-libmaps ros-humble-mrpt-libmath ros-humble-mrpt-libobs ros-humble-mrpt-libopengl
  ros-humble-mrpt-libposes ros-humble-mrpt-libros-bridge
0 upgraded, 7 newly installed, 0 to remove and 42 not upgraded.
Need to get 5154 kB of archives.
After this operation, 19.4 MB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-mrpt-libbase amd64 2.13.7-2jammy.20240825.171436 [1057 kB]
Get:2 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-mrpt-libmath amd64 2.13.7-2jammy.20240825.172230 [851 kB]
Get:3 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-mrpt-libposes amd64 2.13.7-2jammy.20240825.174002 [454 kB]
Get:4 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-mrpt-libopengl amd64 2.13.7-2jammy.20240825.174827 [1252 kB]                
Get:5 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-mrpt-libobs amd64 2.13.7-2jammy.20240825.175621 [703 kB]                    
Get:6 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-mrpt-libmaps amd64 2.13.7-2jammy.20240825.180231 [788 kB]                   
Get:7 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-mrpt-libros-bridge amd64 2.13.7-2jammy.20240825.181046 [49.5 kB]            
Fetched 5154 kB in 9s (552 kB/s)                                                                                                                  
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 123964 files and directories currently installed.)
Preparing to unpack .../0-ros-humble-mrpt-libbase_2.13.7-2jammy.20240825.171436_amd64.deb ...
Unpacking ros-humble-mrpt-libbase (2.13.7-2jammy.20240825.171436) ...
dpkg: error processing archive /tmp/apt-dpkg-install-kEk9W9/0-ros-humble-mrpt-libbase_2.13.7-2jammy.20240825.171436_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/humble/include/mrpt/config/include/mrpt/config/CConfigFile.h', which is also in package ros-humble-mrpt2 2.13.5-1jammy.20240808.122253
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../1-ros-humble-mrpt-libmath_2.13.7-2jammy.20240825.172230_amd64.deb ...
Unpacking ros-humble-mrpt-libmath (2.13.7-2jammy.20240825.172230) ...
dpkg: error processing archive /tmp/apt-dpkg-install-kEk9W9/1-ros-humble-mrpt-libmath_2.13.7-2jammy.20240825.172230_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/humble/include/mrpt/math/include/mrpt/3rdparty/CSparse/cs.h', which is also in package ros-humble-mrpt2 2.13.5-1jammy.20240808.122253
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../2-ros-humble-mrpt-libposes_2.13.7-2jammy.20240825.174002_amd64.deb ...
Unpacking ros-humble-mrpt-libposes (2.13.7-2jammy.20240825.174002) ...
dpkg: error processing archive /tmp/apt-dpkg-install-kEk9W9/2-ros-humble-mrpt-libposes_2.13.7-2jammy.20240825.174002_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/humble/include/mrpt/bayes/include/mrpt/bayes/CKalmanFilterCapable.h', which is also in package ros-humble-mrpt2 2.13.5-1jammy.20240808.122253
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../3-ros-humble-mrpt-libopengl_2.13.7-2jammy.20240825.174827_amd64.deb ...
Unpacking ros-humble-mrpt-libopengl (2.13.7-2jammy.20240825.174827) ...
dpkg: error processing archive /tmp/apt-dpkg-install-kEk9W9/3-ros-humble-mrpt-libopengl_2.13.7-2jammy.20240825.174827_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/humble/include/mrpt/img/include/mrpt/3rdparty/do_opencv_includes.h', which is also in package ros-humble-mrpt2 2.13.5-1jammy.20240808.122253
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../4-ros-humble-mrpt-libobs_2.13.7-2jammy.20240825.175621_amd64.deb ...
Unpacking ros-humble-mrpt-libobs (2.13.7-2jammy.20240825.175621) ...
dpkg: error processing archive /tmp/apt-dpkg-install-kEk9W9/4-ros-humble-mrpt-libobs_2.13.7-2jammy.20240825.175621_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/humble/include/mrpt/obs/include/mrpt/maps/CMetricMap.h', which is also in package ros-humble-mrpt2 2.13.5-1jammy.20240808.122253
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../5-ros-humble-mrpt-libmaps_2.13.7-2jammy.20240825.180231_amd64.deb ...
Unpacking ros-humble-mrpt-libmaps (2.13.7-2jammy.20240825.180231) ...
dpkg: error processing archive /tmp/apt-dpkg-install-kEk9W9/5-ros-humble-mrpt-libmaps_2.13.7-2jammy.20240825.180231_amd64.deb (--unpack):
[... cut ...]
 /tmp/apt-dpkg-install-kEk9W9/4-ros-humble-mrpt-libobs_2.13.7-2jammy.20240825.175621_amd64.deb
 /tmp/apt-dpkg-install-kEk9W9/5-ros-humble-mrpt-libmaps_2.13.7-2jammy.20240825.180231_amd64.deb
 /tmp/apt-dpkg-install-kEk9W9/6-ros-humble-mrpt-libros-bridge_2.13.7-2jammy.20240825.181046_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-humble-mrpt-libros-bridge] failed
  apt: Failed to detect successful installation of [ros-humble-mrpt-libros-bridge]

```

closed 2024-12-25 by jlblancoc