MRPT/mrpt #1329: Running the kf-slam application in the apps folder using the KITTI dataset fails

wqklihbf, 2024-12-07:
[image]
[image]
As shown in the figure above, I have converted the LiDAR data from the KITTI dataset into the rawlog file format required by kf-slam. However, when I run it, I encounter the following issues:
[image]
[image]
Do I need to modify any other configurations?

jlblancoc, 2024-12-09:
Hi, 
Sorry for the misunderstanding, but kf-slam is a proxy to the C++ class:

[mrpt::slam::CRangeBearingKFSLAM](https://docs.mrpt.org/reference/latest/class_mrpt_slam_CRangeBearingKFSLAM.html#doxid-classmrpt-1-1slam-1-1-c-range-bearing-k-f-s-l-a-m)

which, as said in those (brief) docs, implements good-old EKF-SLAM for **landmarks**, that is, features identified by range+attitude angles, and which in MRPT are implemented as [mrpt::obs::CObservationBearingRange](https://docs.mrpt.org/reference/latest/class_mrpt_obs_CObservationBearingRange.html#doxid-classmrpt-1-1obs-1-1-c-observation-bearing-range).

So, to allow EKF-SLAM to run on KITTI, you should design some method to identify key features in each scan and convert them into CObservationBearingRange first. 

If you are looking for point-cloud SLAM for KITTI, the project based on MRPT that implements that is MOLA LiDAR odometry, here: https://docs.mola-slam.org/latest/mola_lo_apps.html
There's a direct CLI command to run it on KITTI sequences.

Cheers

closed 2024-12-09 by jlblancoc

wqklihbf, 2024-12-10:
> 嗨，很抱歉造成误解，但 kf-slam 是 C++ 类的代理：
> 
> [mrpt：：slam：：CRangeBearingKFSLAM](https://docs.mrpt.org/reference/latest/class_mrpt_slam_CRangeBearingKFSLAM.html#doxid-classmrpt-1-1slam-1-1-c-range-bearing-k-f-s-l-a-m)
> 
> 正如那些（简短的）文档所说，它为**地标**实现了老式的 EKF-SLAM，即由范围+姿态角识别的特征，并且在 MRPT 中被实现为 [mrpt：：obs：：CObservationBearingRange](https://docs.mrpt.org/reference/latest/class_mrpt_obs_CObservationBearingRange.html#doxid-classmrpt-1-1obs-1-1-c-observation-bearing-range)。
> 
> 因此，要让 EKF-SLAM 在 KITTI 上运行，您应该设计一些方法来识别每次扫描中的关键特征，并首先将它们转换为 CObservationBearingRange。
> 
> 如果您正在为 KITTI 寻找点云 SLAM，该项目基于 MRPT，实现了 MOLA LiDAR 里程计，请在此处：https://docs.mola-slam.org/latest/mola_lo_apps.html 有一个直接的 CLI 命令可以在 KITTI 序列上运行它。
> 
> 干杯

Thank you for your prompt response!
I've noticed that MOLA uses particle filters, and I'm wondering if there are any similar projects (point-cloud SLAM) that use Extended Kalman Filters?
If not, would it be difficult for me to implement it myself using MRPT? I haven't been working with SLAM for long and am still a beginner.

jlblancoc, 2024-12-10:
There are some Lidar Inertial Odometry (LIO)  libraries out there using EKF, but I can't recall which ones exactly. Look for LIO-SAM, and derived works. 
Most use optimization (Gauss-Newton,...), like LOAM, or MOLA-LO. 

MOLA itself does not use PF. It produces maps that can be used by the ROS2 packages in "mrpt_navigation", which in turn use MRPT inside, for localization with PF, as described [here](https://docs.mola-slam.org/latest/localization.html).